tahvane1/jablotron80 #1: Integration not available in the integrations UI

grisu911, 2021-01-03:
Hi Erkka,

I tryed to use your intergration.
The jablotron is connected via /dev/hidraw0 with a JA-82T

How does the integration work in the ui? Should I find it here?
[image]
If yes than it is not shown.

Or did I misunderstand you?
Can I config the integration any other way?

JohnnyM84, 2021-01-05:
You need to install it manualy in custom components or with hacs

elenril2, 2021-02-23:
I have the same problem. I've installed integration via HACS and can't add it in configuration UI. Is there any option to add it from configuration.yaml?

closed 2021-03-04 by tahvane1