tahvane1/jablotron80 #2: Cannot arm/disarm

JohnnyM84, 2021-01-05:
Hello,
I tried your new integration for jablotron80, was using the old one, thought it maybe useful to have the connected devices in home assistant that can be used.
but this one is not working for me, i have it connected with a JA82-T usb cable to a JA-80K central unit with 2 keypads JA-81F, 3 door sensors, 7 PIRs and 4 smoke alarms, 1 sirene.

I can not arm or disarm it, if i arm it you see on the keypad that he wants to arm it for 1 sec and than stops. The only thing that i see its working is if i arm a zone on the fysical keypad (JA-80K zone A armed) then this goes to on.
Door and PIR sensors do not change in home assistant as detected.

in the home assistant logging i have the following errors:
error 1:

Invalid message of type Settings received e5 0f 04 08 03 3a 03 3a ff
Invalid message of type State received ed 20 42 e8 0d 22 ff
Invalid message of type Settings received e5 12 21 14 00 44 ed 20 42 00 01 00 00 00 2a ff
Invalid message of type StateDetail received e8 e8 0b 53 ff
Invalid message of type State received ed 0b 53 ff
error 2:
Unknown activity received data=ed 53 44 00 0f 04 00 00 6c ff

error 3:

Unknown message type 192 with data c0 00 00 00 00 44 fe ed 20 42 00 01 00 00 00 2a ff received
Unknown message type 202 with data ca 04 00 00 88 fc ed 20 42 00 01 00 00 00 2a ff received
Unknown message type 32 with data 20 42 00 01 00 00 00 2a ff received
Unknown message type 198 with data c6 ff received
Unknown message type 192 with data c0 e8 0e 4b ff received
error 4:
Unexpected error: ‘JA80CentralUnit’ object has no attribute ‘_get_zone’

tahvane1, 2021-03-03:
Hello,
how is your system configured? Partial/Split/Unsplit?

It might be that logic choosing which zone to arm  in alarm_control_panel.py is incorrect... but I have not been able to test it completely...

JohnnyM84, 2021-03-04:
What do you mean with partial/Split/Unsplit?
I have 2 zones, zone a and b. And i use arm away (zone a and b) and arm home (zone a)

I am using Now this plug-in: https://github.com/mattsaxon/HASS-Jablotron80 and that is working. Only it does not add the devices which this plug-in can do.

tahvane1, 2021-03-06:
Hi,
could you try with latest version if it is working okay for you?

Br,
Erkka

JohnnyM84, 2021-03-06:
Tested the new version, updated it with hacs.
It does more that the last version.

If i use arm home, you see on the keypad for 1 sec A lighting up and it beeps, so it wants to arm it but directly dissarms it.
same for arm ayway but ABC is lighting up.

After i few times trying i get managed to arm it A or ABC is lighting up and it beeps that it arms, but after that in the display i get the message complete (it think this is the right translation), after that it arms.

So it looks like everything is going good but there is a wrong input after the arm and then it goes wrong.

Diasarming is working now correctly now.

Errors in log:
`Logger: custom_components.jablotron80
Source: custom_components/jablotron80/jablotron.py:887
Integration: Jablotron 80 (documentation, issues)
First occurred: 10:30:01 (63 occurrences)
Last logged: 11:03:19

Unknown message type 15 with data 0f 06 ff received
Unknown message type 15 with data 0f 33 ff received
Unknown message type 224 with data e0 12 21 14 00 66 ff received
Unknown message type 32 with data 20 42 00 01 00 00 00 2a ff received
Unknown message type 198 with data c6 ff received`

`Logger: custom_components.jablotron80
Source: custom_components/jablotron80/jablotron.py:1585
Integration: Jablotron 80 (documentation, issues)
First occurred: 10:43:52 (10 occurrences)
Last logged: 11:01:14

Unknown activity received data=ed 43 4c 00 0f 04 00 00 18 ff
Unknown activity received data=ed 43 40 00 0f 04 00 00 5c ff
Unknown activity received data=ed 41 4c 00 09 04 00 00 75 ff
Unknown activity received data=ed 41 40 00 09 04 00 00 31 ff`

`Logger: custom_components.jablotron80
Source: custom_components/jablotron80/jablotron.py:1608
Integration: Jablotron 80 (documentation, issues)
First occurred: 10:42:36 (76 occurrences)
Last logged: 11:01:02

Unknown activity received data=ed 53 4c 00 0f 06 00 00 4e ff
Unknown activity received data=ed 51 4c 00 09 00 00 00 6e ff
Unknown activity received data=ed 51 4c 00 09 04 00 00 18 ff
Unknown activity received data=ed 51 4c 00 09 05 00 00 54 ff
Unknown activity received data=ed 51 4c 00 09 06 00 00 23 ff`

`Logger: custom_components.jablotron80
Source: custom_components/jablotron80/jablotron.py:895
Integration: Jablotron 80 (documentation, issues)
First occurred: 10:30:11 (6 occurrences)
Last logged: 10:45:26

Invalid message of type Settings received e5 e8 0d 22 ff
Invalid message of type Settings received e5 12 21 14 00 66 e8 0d 22 ff
Invalid message of type StateDetail received e8 e8 0b e8 8e ff
Invalid message of type Event received e3 81 e3 01 08 03 51 0c 3f 6d ff
Invalid message of type Event received e3 81 e3 01 08 03 54 0d 3f 7d ff`

`Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/jablotron80/jablotron.py:1960
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 10:42:27 (1 occurrences)
Last logged: 10:42:27

[2413751888] can only concatenate str (not "NoneType") to str
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/websocket_api/commands.py", line 135, in handle_call_service
    await hass.services.async_call(
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/core.py", line 1445, in async_call
    task.result()
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/core.py", line 1480, in _execute_service
    await handler.job.target(service_call)
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/helpers/service.py", line 593, in entity_service_call
    future.result()  # pop exception if have
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/helpers/entity.py", line 664, in async_request_call
    await coro
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/helpers/service.py", line 630, in _handle_entity_call
    await result
  File "/home/homeassistant/.homeassistant/custom_components/jablotron80/alarm_control_panel.py", line 122, in async_alarm_arm_home
    self._cu.arm(code,"A")
  File "/home/homeassistant/.homeassistant/custom_components/jablotron80/jablotron.py", line 1960, in arm
    self.send_key_press({"A":"*2","B":"*3","C":"*1"}[zone]+code)
TypeError: can only concatenate str (not "NoneType") to str`

JohnnyM84, 2021-03-06:
Used the olink software for events and it's directly unsetting with the master code.

To answer your previous question how my system is configured.
PGX function: AB set (NOT c) = PG on
PGY function: any system part set = PG on
Partial setting (arming) or system splitting: partial setting (A, AB, ABC)

tahvane1, 2021-03-06:
Hi,

self.send_key_press({"A":"*2","B":"*3","C":"*1"}[zone]+code)
TypeError: can only concatenate str (not "NoneType") to str

This error should come only if "Require code to arm" is set in options of integration and UI doesn't send code.
If  you set this option off then it should use master code automatically. I will anyway fix it that it won't give error.

If "Require code to disarm" is not set in options then it should use master code for disarming.

Br,
Erkka

JohnnyM84, 2021-03-06:
Require code to arm was default on, i did set it to off for testing if it behaved diferently but this was not the case.
Require code to disarm i have on.

Any idea why its directly unsetting after arm?

tahvane1, 2021-03-06:
HA changes status to arming before anything comes from Jablotron so most probably status is not changed at all (fails to arm).
I did some fixes see if anything changed...

JohnnyM84, 2021-03-06:
If i want to add the integration now i get the error:
Aborted
Could not read settings! Is system disarmed?

erross in log:
Logger: custom_components.jablotron80
Source: custom_components/jablotron80/jablotron.py:1916
Integration: Jablotron 80 (documentation, issues)
First occurred: 13:42:31 (1 occurrences)
Last logged: 13:42:31

Trying to enter elevated mode but not reliable status yet

Logger: custom_components.jablotron80.config_flow
Source: custom_components/jablotron80/config_flow.py:101
Integration: Jablotron 80 (documentation, issues)
First occurred: 13:42:36 (1 occurrences)
Last logged: 13:42:36

Could not read settings!

tahvane1, 2021-03-06:
Hi,
what is shown in logs with Debug level? At least in first message you have received settings... also check that system is disarmed before adding...

Br,
Erkka

JohnnyM84, 2021-03-06:
System is disarmed so thats not the problem.

Seeing this in the debug log:
2021-03-06 14:00:28 INFO (MainThread) [custom_components.jablotron80] Init JablotronConnection of type JA_82T with device /dev/hidraw0
2021-03-06 14:00:28 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call system_status
2021-03-06 14:00:28 INFO (MainThread) [custom_components.jablotron80] Connecting to JA80 via JA_82T using /dev/hidraw0...
2021-03-06 14:00:28 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call mode
2021-03-06 14:00:28 INFO (MainThread) [custom_components.jablotron80] JA80CentralUnit(N/A): initializing mode to Unsplit
2021-03-06 14:00:28 INFO (MainThread) [custom_components.jablotron80] initializing
2021-03-06 14:00:28 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Loop endlessly reading serial
2021-03-06 14:00:28 INFO (MainThread) [custom_components.jablotron80] initialization done.

2021-03-06 14:00:33 WARNING (MainThread) [custom_components.jablotron80] Trying to enter elevated mode but not reliable status yet

2021-03-06 14:00:38 ERROR (MainThread) [custom_components.jablotron80.config_flow] Could not read settings!

JohnnyM84, 2021-03-06:
downgraded too 0.11 but same message, if i press a key on the keypad the first press does not do anything the second time it works. looks like the system does not like this testing.
I have too look into it.

tahvane1, 2021-03-06:
Okay final try for today. If still not working I have to continue some other day...

JohnnyM84, 2021-03-06:
for a strange reason i get it only working if i don't disable the old version that i am using.
I have a debug of arm home, here you see that after arming it is disarming directly:

> Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,114 - custom_components.jablotron80 - ERROR - Unknown message type 198 with data c6 ff received
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,115 - custom_components.jablotron80 - DEBUG - Message of type State received ed 40 40 00 01 00 00 00 7d ff
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,115 - custom_components.jablotron80 - DEBUG - JablotronLed function call active
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,116 - custom_components.jablotron80 - DEBUG - JablotronLed function call active
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,117 - custom_components.jablotron80 - DEBUG - JablotronLed function call active
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,117 - custom_components.jablotron80 - DEBUG - JablotronSensor function call value
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,118 - custom_components.jablotron80 - DEBUG - Zone A action disarm
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,118 - custom_components.jablotron80 - DEBUG - JablotronZone function call status
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,119 - custom_components.jablotron80 - DEBUG - Zone B action disarm
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,119 - custom_components.jablotron80 - DEBUG - JablotronZone function call status
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,120 - custom_components.jablotron80 - DEBUG - Zone C action disarm
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02,120 - custom_components.jablotron80 - DEBUG - JablotronZone function call status
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02 ERROR (MainThread) [custom_components.jablotron80] Unknown message type 198 with data c6 ff received
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 40 00 01 00 00 00 7d ff
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
Mar 06 15:31:02 hassbian hass[571]: 2021-03-06 15:31:02 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,127 - custom_components.jablotron80 - DEBUG - Message of type State received ed 40 50 00 01 00 00 00 4f ff
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,130 - custom_components.jablotron80 - DEBUG - JablotronLed function call active
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 50 00 01 00 00 00 4f ff
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,131 - custom_components.jablotron80 - DEBUG - JablotronLed function call active
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,131 - custom_components.jablotron80 - DEBUG - JablotronLed function call active
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,131 - custom_components.jablotron80 - DEBUG - JablotronLed function call active
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,132 - custom_components.jablotron80 - DEBUG - JablotronLed function call active
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,132 - custom_components.jablotron80 - DEBUG - JablotronSensor function call value
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,132 - custom_components.jablotron80 - DEBUG - Zone A action disarm
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,133 - custom_components.jablotron80 - DEBUG - JablotronZone function call status
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,133 - custom_components.jablotron80 - DEBUG - Zone B action disarm
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,133 - custom_components.jablotron80 - DEBUG - JablotronZone function call status
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,133 - custom_components.jablotron80 - DEBUG - Zone C action disarm
Mar 06 15:31:08 hassbian hass[571]: 2021-03-06 15:31:08,134 - custom_components.jablotron80 - DEBUG - JablotronZone function call status
[... cut ...]
Mar 06 15:31:29 hassbian hass[571]: 2021-03-06 15:31:29,238 - custom_components.jablotron80 - DEBUG - Zone C action disarm
Mar 06 15:31:29 hassbian hass[571]: 2021-03-06 15:31:29 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
Mar 06 15:31:29 hassbian hass[571]: 2021-03-06 15:31:29,239 - custom_components.jablotron80 - DEBUG - JablotronZone function call status
Mar 06 15:31:29 hassbian hass[571]: 2021-03-06 15:31:29 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status

mattsaxon, 2022-02-01:
@JohnnyM84 can you try the latest version. I'm the author of the original one and I had to run side by side too until the latest changes. If you can test and confirm, we can close this issue

mattsaxon, 2022-02-01:
@JohnnyM84 which version did you take, did you take the latest commit, or the latest release? @tahvane1 only just cut a new release, so unless you were taking the very latest commit, you may not have had the recent changes.

Sorry to have not made that clear before.

mattsaxon, 2022-02-01:
@JohnnyM84 ignore my last comment if you see it via email as it was incorrect, you did have the release that was needed. Sorry you are still having problems.

JohnnyM84, 2022-02-01:
@mattsaxon it was release 0.16, I can try 0.17

mattsaxon, 2022-02-01:
There's no point unfortunately,  0.16 has the fix that was needing me to run side by side.

JohnnyM84, 2022-02-01:
@mattsaxon i tried adding it again with 0.17 but now with the old integration on, because that worked before i remembered. 
now the configuration succeded. 
I can now arm and disarm zone a (arm home) and zone ab (arm away), so this is solved. 

Bit this is the only thing that works, contact sensors and pir sensors do not work as open/close and detected/not detected.

mattsaxon, 2022-02-01:
When you say they "don't work" are you saying they don't trigger if you walk in front of them when the alarm is set? If the alarm is unset, they will not trigger.

Can you confirm you needed the old component there only during install and you've now removed it, or do you need it there after setup too?

JohnnyM84, 2022-02-01:
@mattsaxon they don’t trigger when te alarm is set. 

I have at the moment both integrations running, will test tomorrow if I turn off the old one if arming still works.

JohnnyM84, 2022-02-02:
@mattsaxon I tested this morning, disabled the old integration in the configuration.yaml and rebooted ha.
After this i can still arm and disarm, so i needed only the old integration for setup.

Tested to trigger an alarm in a delayed zone, than the entities do not change state, if i trigger a zone that is not delayed is see changing that pir to detected, is this the expected behavior or should you see it in a delayed zone too?

mattsaxon, 2022-02-02:
Not what expected behaviour of the sensors is, but feel to me like this is a bug or an enhancement here in addition to your install issues. I'll raise something on this.

Back to install, given you have it working now, would you mind removing the integration and resetting up and seeing if it works without the old component running. If so I think we could close this issue as a one off.

mattsaxon, 2022-02-02:
Can I confirm, this ticket is called cannot arm/disarm, are you still in that situation, or is it better to rename it to "some events not procesed" now?

JohnnyM84, 2022-02-02:
Yeah you can rename it, the arm/disarm problem is solved.

mattsaxon, 2022-02-02:
Great, glad you are a few steps forward! I've got some similar issues with my system, so let me fix those and then we can see what residual ones remain with yours.

JohnnyM84, 2022-02-02:
Ok, if there are some updates i will test them. 
Thanks so far for the updates.

mattsaxon, 2022-02-10:
Fixed a number of issues in https://github.com/tahvane1/jablotron80/pull/35.

mattsaxon, 2022-02-16:
Thanks, I believe I know what the last 4 codes are not, but the "beep" issue I haven't experienced. Do you know when it happens? Is your keypad giving any audible signal?

JohnnyM84, 2022-02-17:
The beep issue occurres every 25 secs, this is the full log:
when disarmed:
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 40 00 01 00 00 00 7d ff
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2022-02-16 20:32:09 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] Message of type Beep received c6 ff
2022-02-16 20:32:34 ERROR (MainThread) [custom_components.jablotron80] Unexpected error:bytearray(b'\xc6\xff'): Traceback (most recent call last):
File "/config/custom_components/jablotron80/jablotron.py", line 2077, in processing_loop
self._process_message(record)
File "/config/custom_components/jablotron80/jablotron.py", line 1954, in _process_message
beep = JablotronKeyPress.get_beep_option(data[0]& 0x0f)
File "/config/custom_components/jablotron80/jablotron.py", line 771, in get_beep_option
return JablotronKeyPress._BEEP_OPTIONS[code]
KeyError: 6
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 40 00 01 00 00 00 7d ff
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2022-02-16 20:32:34 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-16 20:32:59 DEBUG (MainThread) [custom_components.jablotron80] Message of type Beep received c6 ff
2022-02-16 20:32:59 ERROR (MainThread) [custom_components.jablotron80] Unexpected error:bytearray(b'\xc6\xff'): Traceback (most recent call last):
File "/config/custom_components/jablotron80/jablotron.py", line 2077, in processing_loop
self._process_message(record)
File "/config/custom_components/jablotron80/jablotron.py", line 1954, in _process_message
beep = JablotronKeyPress.get_beep_option(data[0]& 0x0f)
File "/config/custom_components/jablotron80/jablotron.py", line 771, in get_beep_option
return JablotronKeyPress._BEEP_OPTIONS[code]
KeyError: 6

When armed home:
2022-02-17 03:18:47 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 41 40 00 09 04 00 00 31 ff
2022-02-17 03:18:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-17 03:18:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-17 03:18:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-17 03:18:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-17 03:18:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-17 03:18:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2022-02-17 03:18:47 DEBUG (MainThread) [custom_components.jablotron80] Zone A action armed
2022-02-17 03:18:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call by
2022-02-17 03:18:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-17 03:18:47 ERROR (MainThread) [custom_components.jablotron80] Unknown activity received data=ed 41 40 00 09 04 00 00 31 ff
2022-02-17 03:19:12 DEBUG (MainThread) [custom_components.jablotron80] Message of type Beep received c6 ff
[... cut ...]
2022-02-17 03:19:36 ERROR (MainThread) [custom_components.jablotron80] Unexpected error:bytearray(b'\xc6\xff'): Traceback (most recent call last):
File "/config/custom_components/jablotron80/jablotron.py", line 2077, in processing_loop
self._process_message(record)
File "/config/custom_components/jablotron80/jablotron.py", line 1954, in _process_message
beep = JablotronKeyPress.get_beep_option(data[0]& 0x0f)
File "/config/custom_components/jablotron80/jablotron.py", line 771, in get_beep_option
return JablotronKeyPress._BEEP_OPTIONS[code]
KeyError: 6

mattsaxon, 2022-02-17:
OK, thanks, I know how to deal with that. I'll sort these in my next commit

mattsaxon, 2022-02-17:
I believe fixed in https://github.com/tahvane1/jablotron80/pull/39

mattsaxon, 2022-02-18:
@JohnnyM84 for the unknown activity messages, I believe you are getting these when you are in 'entering' status. Any idea what they relate to, for example are you triggering a detector, what sort? I can obviously quiten these messages, but I'm trying as much as possible to interpret each message fully

mattsaxon, 2022-02-19:
Thanks @JohnnyM84 can I ask what during these events is displayed on your physical keypad? How are you arminbg/disarming the system during these? fob, rfid, code into the keypad?

JohnnyM84, 2022-02-19:
This is with arming and disarming via home assistant. But with key a or abc it’s the same in the display you see only leaf zone

mattsaxon, 2022-02-19:
OK, understood, I can't see the difference between our systems, but my latest commit has silenced these errors.

Not sure if you want to wait for merge or test prior, the relevant branch is here https://github.com/mattsaxon/jablotron80/tree/Matt-Development-2

JohnnyM84, 2022-02-19:
I will wait for the merge here, otherwise I have to swap af few times

mattsaxon, 2022-02-22:
Fixed in PR https://github.com/tahvane1/jablotron80/pull/51

JohnnyM84, 2022-02-24:
I saw that the PR are in the main branch, via hacs i did a download again and choosed the main branch and not the 0.19 branch.

These error i don't see any more after i did a download again of the main branch a second time, don't know what went wrong the first time:
Unexpected error:bytearray(b'\xc6\xff'): Traceback (most recent call last): File "/config/custom_components/jablotron80/jablotron.py", line 2077, in processing_loop self._process_message(record) File "/config/custom_components/jablotron80/jablotron.py", line 1954, in _process_message beep = JablotronKeyPress.get_beep_option(data[0]& 0x0f) File "/config/custom_components/jablotron80/jablotron.py", line 771, in get_beep_option return JablotronKeyPress._BEEP_OPTIONS[code] KeyError: 6

Entity sensor.zone_a (<class 'custom_components.jablotron80.sensor.JablotronZoneSensorEntity'>) implements device_state_attributes. Please report it to the custom component author.

Unknown activity received data=ed 41

This is the only one i get when armin the system:
Unknown activity received data=ed 51
Unknown activity received data=ed 53

So looking good, now the log is not getting as big as with the last version.

mattsaxon, 2022-02-24:
This doesn't look like you have the new code to me as https://github.com/tahvane1/jablotron80/pull/49 fixes the device state attributes,

mattsaxon, 2022-02-24:
However if it is, then we need more data on your unknown activities, please post the full message, not just first 2 bytes.

tahvane1, 2022-02-24:
new release is available

JohnnyM84, 2022-02-24:
Updated to 0.20 but same as the main branch, @mattsaxon i think you didn’t read my post right the first 3 errors are away including your mentioned State attribute. 

Here by the log of the unknown activities (i see 1 activity recognised starting with 51) and sometimes with arming it sounds like the system armes again when already arming (fast beeps at the and getting slow again):
2022-02-24 20:49:15 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2022-02-24 20:49:15 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2022-02-24 20:49:15 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Sending new command Command name=keypress *HIDDEN*
2022-02-24 20:49:15 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Command sent b'\x00\x02\x01\x8f'
2022-02-24 20:49:15 INFO (ThreadPoolExecutor-0_0) [custom_components.jablotron80] confirmation for command Command name=keypress *HIDDEN* received
2022-02-24 20:49:16 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Sending new command Command name=keypress *HIDDEN*
2022-02-24 20:49:16 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Command sent b'\x00\x02\x01\x82'
2022-02-24 20:49:16 INFO (ThreadPoolExecutor-0_0) [custom_components.jablotron80] confirmation for command Command name=keypress *HIDDEN* received
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] Message of type KeyPress received 8f ff
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] Message of type Beep received a0 ff
2022-02-24 20:49:16 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa0, 1 subtle (short) beep triggered
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 [44 00 01 00 00 00 20](tel:44 00 01 00 00 00 20) ff
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-24 20:49:16 DEBUG (MainThread) [custom_components.jablotron80] Message of type KeyPress received 82 ff
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] Message of type Beep received a1 ff
2022-02-24 20:49:17 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa1, 1 loud (long) beep triggered
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 51 4c [00 09 00 00 00](tel:00 09 00 00 00) 6e ff
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-24 20:49:17 INFO (MainThread) [custom_components.jablotron80] JablotronLed(1): active changed from False to True
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] Zone A action arming
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call by
2022-02-24 20:49:17 INFO (MainThread) [custom_components.jablotron80] JablotronZone(1): initializing by to Device id=0,model=JA-80K,reaction=None,tampered=False,battery_low=Falsename=JA-80K Control Panel,active=False
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-24 20:49:17 INFO (MainThread) [custom_components.jablotron80] JablotronZone(1): status changed from Disarmed to Arming
2022-02-24 20:49:17 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2022-02-24 20:49:17 INFO (MainThread) [custom_components.jablotron80] Zone A status changed from Disarmed to Arming
[... cut ...]
2022-02-24 20:50:06 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-24 20:50:06 INFO (MainThread) [custom_components.jablotron80] JablotronZone(1): status changed from Arming to Armed
2022-02-24 20:50:06 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2022-02-24 20:50:06 INFO (MainThread) [custom_components.jablotron80] Zone A status changed from Arming to Armed
2022-02-24 20:50:07 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 41 [40 00 09 04 00 00 31](tel:40 00 09 04 00 00 31) ff

mattsaxon, 2022-02-24:
OK, sorry I did misread. Watch this space, more fixes coming, but they are all a bit entangled so you might have to wait a bit for various reviews and merges.

Once all my changes are in, we can see where we are.

mattsaxon, 2022-02-25:
I believe this is now fixed in the latest code on "main"

JohnnyM84, 2022-02-25:
Tested 0.21, all errors are now away from the ha log.
Other things that i neef to test? Notised 1 thing but Will open a new ticket for it.

tahvane1, 2022-02-25:
Let's close this now

closed 2022-02-25 by tahvane1