tahvane1/jablotron80 #3: /dev/ttyUSB1 not accepeted

tinus20vt, 2021-02-27:
Hi,

I'm trying to get it work but the plugin doesn't accept my path to the USB-cable.
When i fill in "/dev/ttyUSB1" and click on save i get the error "Unknown error occurred" directly.
[image]

When i try the path of the cable it also not working.
[image]

If i leave it as it is, i can continu but then i get the error "could not read setting" (off course)
Do you have a solution?

Thanks in advance

tahvane1, 2021-03-03:
What cable are you using? I have tested only JA-82T

tinus20vt, 2021-03-03:
Thanks for your reply, i have the JA-80T and HA is assigning it to ttyUSB1

tahvane1, 2021-03-03:
Hi,
you could try this
In Jablotron.py insert in the beginning
import serial

and then change line 580 (in connect method)
self._connection = serial.Serial(
                port=self.device,
                baudrate=9600,
                parity=serial.PARITY_NONE,
                bytesize=serial.EIGHTBITS,
                dsrdtr=True,
                # stopbits=serial.STOPBITS_ONE
                timeout=1)

and see what happens

tinus20vt, 2021-03-03:
Hi Thanks for your help.

I've got the same error, i tried your code in 3 ways but all with the same
result "unknow error occurred"
1st:
[image: image.png]
2nd:
[image: image.png]
last:
[image: image.png]

kind regard,

Martijn

tahvane1, 2021-03-04:
Hi,
problem might be that data is different. In JA-82T  first data is
(jablotron.py line 627)
			if  data[0] == 0x82:

maybe you could try to replace this
```
def _read_data(self, max_package_sections: int =15)->List[bytearray]:
		read_buffer = []
		ret_val = []
		for i in range(max_package_sections):
			data = self._connection.read(64)
		   # if LOGGER.isEnabledFor(logging.DEBUG):
		   #     formatted_data = " ".join(["%02x" % c for c in data])
		   #     LOGGER.debug(f'Received raw data {formatted_data}')
			if  data[0] == 0x82:
				size = data[1] 
				read_buffer.append(data[2:2+int(size)])
				if data[1 + int(size)] == 0xff:
					# return data received
					ret_bytes = []
					for i in b''.join(read_buffer):
						ret_bytes.append(i)
						if i == 0xff:
							ret_val.append(bytearray(ret_bytes))
							ret_bytes.clear()
					return ret_val
		return ret_val
```

with this
```
def _read_data(self, max_package_sections: int =15)->List[bytearray]:
		read_buffer = []
		ret_val = []
		for i in range(max_package_sections):
			data = self._connection.read(64)
		        if LOGGER.isEnabledFor(logging.DEBUG):
		            formatted_data = " ".join(["%02x" % c for c in data])
		            LOGGER.debug(f'Received raw data {formatted_data}')
			
	        ret_bytes = []
	        for i in b''.join(read_buffer):
		        ret_bytes.append(i)
		        if i == 0xff:
			        ret_val.append(bytearray(ret_bytes))
			        ret_bytes.clear()
	        return ret_val
```
If it is not working straight away then at least you should see from logs what you are receiving

tinus20vt, 2021-03-05:
Thanks for all your effort.
Unfortunately, when i change the lines with the new code, i get directly an
error when i want to add the integration.

"Error, configflow can't be loaded"

Logger: aiohttp.server
Source: custom_components/jablotron80/__init__.py:12
First occurred: 13:13:41 (2 occurrences)
Last logged: 13:14:40
Error handling request

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py",
line 422, in _handle_request
    resp = await self._request_handler(request)
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_app.py",
line 499, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_middlewares.py",
line 119, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py",
line 56, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py",
line 18, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py",
line 72, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py",
line 129, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py",
line 129, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py",
line 117, in post
    return await super().post(request)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py",
line 60, in wrapper
    result = await method(view, request, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py",
line 69, in post
    result = await self._flow_mgr.async_init(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line
122, in async_init
    flow = await self.async_create_flow(handler, context=context, data=data)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line
561, in async_create_flow
    integration.get_platform("config_flow")
  File "/usr/src/homeassistant/homeassistant/loader.py", line 499, in
get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 504, in
_import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/local/lib/python3.8/importlib/__init__.py", line 127, in
import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/config/custom_components/jablotron80/__init__.py", line 12, in <module>
    from .jablotron import JA80CentralUnit
  File "/config/custom_components/jablotron80/jablotron.py", line 640
    def _read_data(self, max_package_sections: int =15)->List[bytearray]:
                                                                        ^
IndentationError: unindent does not match any outer indentation level

tahvane1, 2021-03-05:
Hi,
I updated code in such way that it might just work. It is not in release so you have to test it from git files.

Br,
Erkka

tinus20vt, 2021-03-06:
Hi Erkka,

Thanks for the update.
I removed the old integration and loaded the new one, after that I selected
the JA-80T as cable.
The next message is displayed:
[image: image.png]
In the logs, i get 2 messages:
[image: image.png]

first:
[image: image.png]

second:
[image: image.png]

Martijn

tahvane1, 2021-03-06:
Hi Martin,
Images were not visible. Could you repost content?
Br,
Erkka

tinus20vt, 2021-03-06:
> Hi Martin,
> Images were not visible. Could you repost content?
> Br,
> Erkka

Done

tahvane1, 2021-03-06:
Hi,
can you post a bit of input from 

hexdump /dev/ttyUSB1

run in console or some lines of output log (of RAW data) as advices in README
Br,
Erkka

tinus20vt, 2021-03-06:
```
2021-03-06 10:21:09 INFO (MainThread) [custom_components.jablotron80] Init JablotronConnection of type JA_80T with device /dev/ttyUSB0
2021-03-06 10:21:09 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call system_status
2021-03-06 10:21:09 INFO (MainThread) [custom_components.jablotron80] Connecting to JA80 via JA_80T using /dev/ttyUSB0...
2021-03-06 10:21:09 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call mode
2021-03-06 10:21:09 INFO (MainThread) [custom_components.jablotron80] JA80CentralUnit(N/A): initializing mode to Unsplit
2021-03-06 10:21:09 INFO (MainThread) [custom_components.jablotron80] initializing
2021-03-06 10:21:09 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Loop endlessly reading serial
2021-03-06 10:21:09 INFO (MainThread) [custom_components.jablotron80] initialization done.
2021-03-06 10:21:10 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Received raw data ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff
2021-03-06 10:21:11 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Received raw data ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff
2021-03-06 10:21:12 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Received raw data ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff
2021-03-06 10:21:14 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Received raw data ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff
2021-03-06 10:21:14 WARNING (MainThread) [custom_components.jablotron80] Trying to enter elevated mode but not reliable status yet
2021-03-06 10:21:15 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Received raw data ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff
2021-03-06 10:21:16 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Received raw data ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff
2021-03-06 10:21:17 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Received raw data ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff
2021-03-06 10:21:19 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Received raw data ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff
2021-03-06 10:21:19 ERROR (MainThread) [custom_components.jablotron80.config_flow] Could not read settings!
2021-03-06 10:21:20 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Received raw data ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff ed 60 16 00 01 00 00 00 7f ff
2021-03-06 10:21:20 INFO (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Disconnecting from JA80...```

tahvane1, 2021-03-06:
Hi,
I have not been able to replicate this (by using your raw data). Are you using master code when setting up integration?
Current code seems to parse input in format your cable provides okay...

Br,
Erkka

tahvane1, 2021-03-06:
Can you try again with latest version?

tinus20vt, 2021-03-06:
I've got an endless loop:
[image]

In my logs (and it keeps going)
[image]
and:
```2021-03-06 15:20:53 INFO (MainThread) [custom_components.jablotron80] Init JablotronConnection of type JA_80T with device /dev/ttyUSB0
2021-03-06 15:20:53 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call system_status
2021-03-06 15:20:53 INFO (MainThread) [custom_components.jablotron80] Connecting to JA80 via JA_80T using /dev/ttyUSB0...
2021-03-06 15:20:53 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call mode
2021-03-06 15:20:53 INFO (MainThread) [custom_components.jablotron80] JA80CentralUnit(N/A): initializing mode to Unsplit
2021-03-06 15:20:53 INFO (MainThread) [custom_components.jablotron80] initializing
2021-03-06 15:20:53 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Loop endlessly reading serial
2021-03-06 15:20:53 INFO (MainThread) [custom_components.jablotron80] initialization done.
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 60 16 00 01 00 00 00 7f ff
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:20:54 INFO (MainThread) [custom_components.jablotron80] JablotronLed(5): active changed from False to True
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-06 15:20:54 INFO (MainThread) [custom_components.jablotron80] JablotronZone(3): status changed from Unknown to Disarmed
2021-03-06 15:20:54 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-03-06 15:20:54 INFO (MainThread) [custom_components.jablotron80] Zone C status changed from Unknown to Disarmed
2021-03-06 15:20:56 ERROR (MainThread) [custom_components.jablotron80] Unknown message type 0 with data 00 7f ff received
2021-03-06 15:20:56 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 60 16 00 01 00 00 00 7f ff
2021-03-06 15:20:56 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:20:56 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:20:56 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:20:56 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:20:56 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:20:56 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-03-06 15:20:56 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-03-06 15:20:56 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-03-06 15:20:56 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-03-06 15:20:56 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-06 15:20:58 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=Elevated mode first part
2021-03-06 15:20:58 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=Elevated mode second part
2021-03-06 15:20:58 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2021-03-06 15:20:58 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2021-03-06 15:20:58 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2021-03-06 15:20:58 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2021-03-06 15:20:58 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=Get settings
2021-03-06 15:20:59 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Sending new command Command name=Elevated mode first part
2021-03-06 15:20:59 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Command sent b'\x00\x02\x01\x8f'
2021-03-06 15:21:00 ERROR (MainThread) [custom_components.jablotron80] Unknown message type 0 with data 00 02 01 8f ff received
2021-03-06 15:21:00 DEBUG (MainThread) [custom_components.jablotron80] Message of type Beep received a0 ff
2021-03-06 15:21:00 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 60 04 00 01 00 00 00 32 ff
2021-03-06 15:21:00 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
[... cut ...]
2021-03-06 15:21:02 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:21:02 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:21:02 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-06 15:21:02 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-03-06 15:21:02 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 125.0 to 75.0```
And still growing ;)

tinus20vt, 2021-03-06:
> Hi,
> I have not been able to replicate this (by using your raw data). Are you using master code when setting up integration?
> Current code seems to parse input in format your cable provides okay...
> 
> Br,
> Erkka

Yes, im using the master code. With ohter codes i get others errors.

tahvane1, 2021-03-06:
So it is working now? Can issue be closed?

tinus20vt, 2021-03-06:
No, unfortunately it isn't working.
When i add the values and click on save then the integration will not start, is going in a loop.
At that moment there is an error report as the picture above is showing. (and the logfiles)

Maybe it isn't compatible with the JA-80T and i should buy a JA-82T.

But thanks for trying.

tahvane1, 2021-03-06:
Okay. So you only get these ed 60 04 00 01 00 00 00 32 ff and nothing else start to come to logs? It might be that settings have to asked in some other way than with JA82-T.  This configuration takes sometime ~ 30 sec to minute. It does same thing as OLink when reading settings. You could check packets moving when you load settings with Olink and I could check if those looks similar to what I have.

I can see now that log was from configuration stage. I thought it was from running (and for that it looked "normal").

tinus20vt, 2021-03-06:
i tried to add the integration again, but after a few minutes (the wheel was spinning):
[image]
and the log (the last part from 2599 lines):
[image]
[image]

I have Olink but can i get a logfile from there?

tahvane1, 2021-03-06:
Under central menu is Packets selection. It will open window that shows communication. Open that and and copy & paste results from the start of opening settings until settings have been completely opened.

tinus20vt, 2021-03-08:
Here you go:

```
ED 60 16 00 01 00 00 28 1C FF 
8F FF 
A0 FF 
80 FF 
A0 FF 
88 FF 
ED 60 04 00 01 00 00 00 32 FF 
A0 FF 
80 FF 
A0 FF 
88 FF 
A0 FF 
80 FF 
A1 FF 
E8 0E 4B FF 
B8 FF 
E3 08 03 13 53 41 7F 79 FF 
E8 0C 64 FF 
ED 00 01 00 01 00 00 00 29 FF 
E8 0C 64 FF 
B7 FF 
8A FF 
A1 FF 
E5 0E 00 09 02 3A FF 
E6 03 00 01 67 FF 
E6 03 01 01 4D FF 
E6 03 02 00 75 FF 
E6 03 03 01 19 FF 
B7 FF 
E6 03 04 00 2A FF 
E6 03 05 01 46 FF 
E6 03 06 00 7E FF 
E6 03 07 01 12 FF 
E6 03 08 01 71 FF 
E6 03 09 00 1D FF 
E6 02 05 00 4C FF 
E6 02 06 00 32 FF 
E6 02 07 00 18 FF 
E6 02 08 01 3D FF 
E6 02 00 02 42 FF 
E6 02 01 04 19 FF 
E6 02 02 02 16 FF 
E6 02 03 07 24 FF 
E6 02 04 01 20 FF 
ED 05 01 00 01 00 00 1E 52 FF 
E6 06 00 00 01 04 09 02 06 0A 05 3B FF 
E6 06 00 00 02 0F 0F 00 00 00 00 4C FF 
E6 06 00 00 03 0F 0F 00 00 00 00 05 FF 
E6 06 00 00 04 0F 0F 00 00 00 00 1F FF 
E6 06 00 00 05 04 09 00 0C 0E 0B 7C FF 
E6 06 00 00 06 01 01 0D 08 05 0C 6F FF 
E6 06 00 00 07 01 01 0D 0F 0B 0D 28 FF 
ED 05 01 00 01 00 00 00 02 FF 
E6 06 00 00 08 05 09 05 0A 05 05 67 FF 
E6 06 00 00 09 05 09 0D 0D 03 03 72 FF 
E6 06 00 01 00 06 04 0F 0A 03 09 0C FF 
E6 06 00 01 01 06 04 0F 07 0B 0D 78 FF 
E6 06 00 01 02 00 00 00 00 00 00 41 FF 
E6 06 00 01 03 05 01 0E 02 0F 0B 34 FF 
E6 06 00 01 04 01 0D 0B 0B 07 0B 5C FF 
E6 06 00 01 05 01 0D 0D 02 06 0C 57 FF 
ED 05 01 00 01 00 00 00 02 FF 
E6 06 00 01 06 0F 0F 00 00 00 00 2C FF 
ED 05 01 00 01 00 00 00 02 FF 
E6 06 00 01 07 0F 0F 00 00 00 00 65 FF 
E6 06 00 01 08 0F 0F 00 00 00 00 18 FF 
E6 06 00 01 09 0F 0F 00 00 00 00 51 FF 
[... 237 lines cut ...]
ED 00 01 00 01 00 00 00 29 FF 
84 FF 
A0 FF 
81 FF 
A0 FF 
83 FF 
A0 FF 
85 FF 
A0 FF 
81 FF 
A0 FF 
80 FF 
A0 FF 
88 FF 
ED 03 04 00 01 00 00 00 62 FF 
A0 FF 
80 FF 
A0 FF 
83 FF 
A0 FF 
82 FF 
A0 FF 
81 FF 
A1 FF 
```

closed 2021-03-17 by tinus20vt