tahvane1/jablotron80 #4: Cannot set integration - Unknown message type 239

sochorf, 2021-02-28:
Hi,

thanks for this nice Jablotron integration.
I installed package via HASC. During integration settings attempt I got error with following debug log (just last part of log):

2021-02-28 16:13:42 ERROR (MainThread) [custom_components.jablotron80] Unknown message type 239 with data ef 42 00 00 00 00 00 00 00 1b ff received
2021-02-28 16:13:42 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 14 01 11 00 00 00 32 ff
2021-02-28 16:13:42 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-02-28 16:13:42 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-02-28 16:13:42 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-02-28 16:13:42 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-02-28 16:13:42 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-02-28 16:13:42 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-02-28 16:13:44 INFO (ThreadPoolExecutor-1_0) [custom_components.jablotron80] no confirmation for command Command name=Get settings received
2021-02-28 16:13:45 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=Esc / back
2021-02-28 16:13:45 ERROR (MainThread) [custom_components.jablotron80] Unknown message type 239 with data ef 42 00 00 00 00 00 00 00 1b ff received
2021-02-28 16:13:45 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 14 01 11 00 00 00 32 ff
2021-02-28 16:13:45 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-02-28 16:13:45 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-02-28 16:13:45 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-02-28 16:13:45 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-02-28 16:13:45 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-02-28 16:13:45 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-02-28 16:13:45 DEBUG (ThreadPoolExecutor-1_0) [custom_components.jablotron80] Sending new command Command name=Esc / back
2021-02-28 16:13:45 DEBUG (ThreadPoolExecutor-1_0) [custom_components.jablotron80] Command sent b'\x00\x02\x01\x8e'
2021-02-28 16:13:45 INFO (ThreadPoolExecutor-1_0) [custom_components.jablotron80] confirmation for command Command name=Esc / back received
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] Message of type KeyPress received 8e ff
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] Message of type Beep received a0 ff
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 10 02 01 00 00 00 38 ff
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-02-28 16:13:46 INFO (MainThread) [custom_components.jablotron80] JablotronLed(4): active changed from True to False
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] JablotronDevice function call active
2021-02-28 16:13:46 INFO (MainThread) [custom_components.jablotron80] JablotronDevice(2): active changed from False to True
2021-02-28 16:13:46 DEBUG (MainThread) [custom_components.jablotron80] Zone C action device_activated
2021-02-28 16:13:47 ERROR (MainThread) [custom_components.jablotron80] Unknown message type 239 with data ef 42 00 00 00 00 00 00 00 1b ff received
2021-02-28 16:13:47 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 10 02 01 00 00 00 38 ff
2021-02-28 16:13:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-02-28 16:13:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-02-28 16:13:47 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-02-28 16:13:47 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-02-28 16:13:47 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-02-28 16:13:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-02-28 16:13:47 DEBUG (MainThread) [custom_components.jablotron80] JablotronDevice function call active
2021-02-28 16:13:47 DEBUG (MainThread) [custom_components.jablotron80] Zone C action device_activated
2021-02-28 16:13:48 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 16 00 01 00 00 00 06 ff
2021-02-28 16:13:48 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-02-28 16:13:48 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-02-28 16:13:48 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-02-28 16:13:48 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-02-28 16:13:48 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-02-28 16:13:48 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-02-28 16:13:49 ERROR (MainThread) [custom_components.jablotron80] Unknown message type 239 with data ef 42 00 00 00 00 00 00 00 1b ff received
2021-02-28 16:13:49 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 16 00 01 00 00 00 06 ff
2021-02-28 16:13:49 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-02-28 16:13:49 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-02-28 16:13:49 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-02-28 16:13:49 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-02-28 16:13:49 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-02-28 16:13:49 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-02-28 16:13:50 ERROR (MainThread) [custom_components.jablotron80.config_flow] Could not read settings!
2021-02-28 16:13:50 INFO (ThreadPoolExecutor-1_0) [custom_components.jablotron80] Disconnecting from JA80...

Thanks and Best Regards
Filip

tahvane1, 2021-03-03:
Hi,
I don't know how CRC code is calculated and thus partial records are received (but not recognized as such).  Try to remove and add configuration again.
Br,
Erkka

tahvane1, 2021-03-06:
It would be nice to know what happens in system this is received ef 42 00 00 00 00 00 00 00 1b ff . I have not seen records with ef before... It looks like "normal" status record though...

sochorf, 2021-03-06:
Hi Erkka,

thanks for your message. I tried to reconfigure the integration for three times. Each attempt with the same result. Integration can't be established. 

With the 0xEF - I have the suspicion on my GSM module [JA-82Y](https://www.jablotron.com/en/about-jablotron/downloads/?filename=ja-82y_en_mld51112.pdf&do=downloadFile). 
I have the Jablotron device on my cottage, I will try disconnect GSM module and run HA integration when I'll be on the cottage again.

Best Regards
Filip

mattsaxon, 2022-02-01:
@sochorf its been a long time since you posted here, is this still an issue and are you still interested to get this working?

Can you try the latest version and see if the issues are the same?