tahvane1/jablotron80 #5: False motion detection

elenril2, 2021-03-09:
Integration detect false motion in one of my sensors (on channel 6) at 19:28:50 and the motion status has been kept on the sensor until now. Alarm was disarmed and there was no maintenance or service mode thus PIR should not react to motion.

```2021-03-08 19:28:49 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-03-08 19:28:49 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-03-08 19:28:49 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-08 19:28:49 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-03-08 19:28:49 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-08 19:28:49 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-03-08 19:28:49 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 00 00 01 00 00 00 16 ff
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-03-08 19:28:50 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 150.0 to 0.0
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 00 00 01 00 00 3c 15 ff
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-03-08 19:28:50 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 0.0 to 150.0
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 00 00 01 00 00 00 16 ff
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-03-08 19:28:50 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 150.0 to 0.0
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-03-08 19:28:50 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status```

elenril2, 2021-03-11:
After couple of days sensor has been deactivated:
```2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronDevice function call active
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] Zone C action device_activated
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 07 06 11 00 00 00 3b ff
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-03-11 01:29:57 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 125.0 to 0.0
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] JablotronDevice function call active
2021-03-11 01:29:57 DEBUG (MainThread) [custom_components.jablotron80] Zone C action device_activated
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-03-11 01:30:18 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 0.0 to 125.0
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] JablotronDevice function call active
2021-03-11 01:30:18 DEBUG (MainThread) [custom_components.jablotron80] Zone C action device_activated
2021-03-11 01:30:19 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 07 06 11 00 00 00 3b ff
2021-03-11 01:30:19 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:19 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:19 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:19 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:19 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:19 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-03-11 01:30:19 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 125.0 to 0.0
2021-03-11 01:30:19 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-03-11 01:30:19 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
[... cut ...]
2021-03-11 01:30:43 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:43 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:43 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:43 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-03-11 01:30:43 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value```

mattsaxon, 2022-02-01:
Is this an issue or can we close?

elenril2, 2022-02-01:
Yes, I think the case was faulty tamper alarm in my device.

tahvane1, 2022-02-01:
Excellent work Matt

closed 2022-02-01 by tahvane1