tahvane1/jablotron80 #6: Error adding integration

tinus20vt, 2021-03-17:
Hi Erkka,

I've closed the old issue, we couldn't get it working with a JA-80T.
So, i bought a JA-82T and i was hoping that this work instantly........

I get an error when add the integration with the following parameters:
[image]

I've added the setting below:
[image]

which give the error:
[image]

The log in HA:
[image]

And the extensive logging:
[image]

Do you have any idea?

tahvane1, 2021-03-17:
Hi,
most probably wrong device. Do you have /dev/hidraw0 or something like that? It should be HID interface (as far as I know)... 
Br,
Erkka

tinus20vt, 2021-03-17:
I thought also. 
When i check my hardware I've:
/dev/usb/hiddev0
/dev/hidraw1
/dev/hidraw0

None of them works and gives the same error.
It is strange that the address from the JA-82 is "DEVNAME: /dev/bus/usb/002/002" never seen before.
I just checked if the JA-82T works but with Olink it just works.

tinus20vt, 2021-03-17:
It is hidraw1:
[image]

But the result:
[image]

tinus20vt, 2021-03-17:
I don't know what happen, first it didn't work. 
So tried the integration from mattsaxon and that works. After that, i tried again with your integration and set the same credentials as before and now it works.

Thanks for all the effort.

closed 2021-03-17 by tinus20vt