tahvane1/jablotron80 #7: "Could not read settings! Is system disarmed?" Error

freschism, 2021-03-26:
Hi there!

I'm trying to set up my JA80 with an 82 connector, using this integration.
Im using HASS on a Virtual Box.

Once I enable the integration, I cannot get past the error "Aborted. Could not read settings! Is system disarmed?"

My theory is that I'm not using the correct /dev/ usb path, however I feel like I've tried every path imagineable.

This is what my device looks like in the Hardware Info:

`  - name: 1-2
    sysfs: '/sys/devices/pci0000:00/0000:00:06.0/usb1/1-2'
    dev_path: /dev/bus/usb/001/003
    subsystem: usb
    by_id: null
    attributes:
      BUSNUM: '001'
      DEVNAME: /dev/bus/usb/001/003
      DEVNUM: '003'
      DEVPATH: '/devices/pci0000:00/0000:00:06.0/usb1/1-2'
      DEVTYPE: usb_device
      DRIVER: usb
      ID_BUS: usb
      ID_FOR_SEAT: usb-pci-0000_00_06_0-usb-0_2
      ID_MODEL: JA-82T_PC_Interface
      ID_MODEL_ENC: JA-82T\x20PC\x20Interface
      ID_MODEL_ID: '0007'
      ID_PATH: 'pci-0000:00:06.0-usb-0:2'
      ID_PATH_TAG: pci-0000_00_06_0-usb-0_2
      ID_REVISION: '0100'
      ID_SERIAL: JABLOTRON_ALARMS_JA-82T_PC_Interface_00000000
      ID_SERIAL_SHORT: '00000000'
      ID_USB_INTERFACES: ':030000:'
      ID_VENDOR: JABLOTRON_ALARMS
      ID_VENDOR_ENC: JABLOTRON\x20ALARMS
      ID_VENDOR_ID: 16d6
      MAJOR: '189'
      MINOR: '2'
      PRODUCT: 16d6/7/100
      SUBSYSTEM: usb
      TAGS: ':seat:'
      TYPE: 0/0/0
      USEC_INITIALIZED: '2860670'`
      
      
[image]
And the above is what my Log says.

Any idea what to do from here?

tahvane1, 2021-05-19:
I think you are trying to use too low level usb. Device should be something like /dev/hidraw0

freschism, 2021-05-19:
I actually managed to get it working! 
Using another machine did the trick.

However, I have another problem - I simply don't know how to get my Motion Sensors (devices) to appear or be detected in Home Assistant. Any guidance here?

tahvane1, 2021-05-19:
You don't have any devices visible or just motion detectors? I personally do not have motion detector connected to Jablotron (but KNX instead) so I have not been able to reverse engineer message protocol for that specific use case... It is kind of implemented but it assumes that it works same way as door detectors...

freschism, 2021-05-19:
Right now I simply don't know how to add them. 
I tried this (Sensor1-19), but HASS doesn't detect any. I wouldn't even know how HASS knows the names of these sensors.
[image]

tahvane1, 2021-05-19:
When you add integration it will show you configured devices. You can also find those under jablotron80 integration.

mazzy89, 2021-05-20:
I'm having the same issue here. after I've cleaned up the integration I could not add it anymore and I get the same error.

i've tried everything. reboot, delete the integration, uninstall entire HomeAssistantOS. nothing helped.

mazzy89, 2021-05-20:
I've fixed this situation in a very funny way. I had to reinstall and configure the old legacy Jablotron automation https://github.com/mattsaxon/HASS-Jablotron80 make it work, arm the system, uninstall it and then this integration worked. very weird.

elenril2, 2021-05-24:
Thank you very much for the tip, it also worked for me after adding the old integration.
In desperation, I also tried almost everything, I was about to reinstall Linux or change the machine...

mazzy89, 2021-05-24:
@elenril2 no problem. A pleasure to help here. My gut feeling is that the system get into a not recognized state which is instead recognized by the old integration that when activated put the state back to an healthy one.

renzo-massaro, 2021-09-19:
Hi all, I use the Jablotron 80 with the 82T cable to my raspberry, all is good but when adding this component I also get the error Could not read settings is system disarmed.

My cable is on /dev/hidraw0 but i dont get it connected. 
Is there some issue with raspberry maybe? The alarm is nog armed when I try to connect.

mazzy89, 2021-09-19:
Follow this workaround https://github.com/tahvane1/jablotron80/issues/7#issuecomment-844927508 so you will get your system out of that bad state

renzo-massaro, 2021-09-19:
When I arm the system with the other component and I uninstall it again, I have to install the other component in armed state or disarmed state then?

mazzy89, 2021-09-19:
Disarmed state

mazzy89, 2021-09-19:
To be more precise in order to install this component correctly the system must get into a service mode which is possibile only in disarmed state.

renzo-massaro, 2021-09-19:
I already set the system in service mode, but without installing the other
component you refer to.. I let you know later if it helped.. thanks for the
very quick response🙂

mazzy89, 2021-09-19:
You do not need to set the system in service mode.

renzo-massaro, 2021-09-19:
Everything is working now indeed after your suggestion, thanks ;)

mazzy89, 2021-09-19:
Niceeeeee. One day I will have to contribute to solve this issue on this component because it's so annoying. The problem is that I need to replicate that and in order to do so I need to beak temporarily the integration home which is used constantly by family. 
But anyways glad it worked.

mazzy89, 2021-09-19:
Oh I forgot to mention one little detail. I'm not sure how you are running home assistant but in my case if I unplug the 82T or reboot the hassio the only way to make it work again is to do this dance over and over again. Which is the thing that annoy me the most. So just be aware of that. @tahvane1 do you think we can prioritize to solve this issue anyhow?

elenril2, 2021-09-20:
I'm running two components simultaneously and mostly don't have a problem. Only once I have lost connection to Jablotron - unfortunately at the exact time when I was leaving my home on the way to the vacation.

mazzy89, 2021-09-20:
@elenril2 why are you running two components simultaneously? In order to avoid the issue discussed here?

elenril2, 2021-09-20:
> @elenril2 why are you running two components simultaneously? In order to avoid the issue discussed here?

Yes, and I'dont have a problem when I reboot HA. I didn't try to disconnect the cable.
In my case I think that this problem occured only when I upgraded HA to newer version and the old integration hadn't version entry in the manifest.

mazzy89, 2021-09-20:
That's interesting. That's a good workaround too to maintain the integration working. I will definitely give a try. Thanks for the idea.

mazzy89, 2021-09-20:
In your case @elenril2 the entities names coming from the two integrations have same name or different names set in the configuration?

elenril2, 2021-09-20:
Different, and the older one is in the read-only state.

mazzy89, 2021-09-20:
@elenril2 how is possible to set up the integration in read-only state? I haven't found any toggle to do that/

elenril2, 2021-09-20:
Hmm... I thought this state was default. My yaml entry looks like this:
```
alarm_control_panel:
  - platform: jablotron
    serial_port: /dev/hidraw0     
    code: !secret jablotron
    code_panel_arm_required: False
    code_panel_disarm_required: True
    code_arm_required: False
    code_disarm_required: False
    sensor_names: 
      1: "Front door"
      2: "Garden door"
      3: "Bedroom"
```
Sensor names are left as in integration example.
And the entity looks like this:
[image]

mazzy89, 2021-09-20:
oh yeah gotcha. that is the default state when the config is managed via the YAML and not via the UI. you meant that.

tahvane1, 2022-01-31:
This is probably related to #16

mattsaxon, 2022-02-01:
@freschism can you try latest version which should not need my old component running at same time. If you can confirm, we can close this issue

freschism, 2022-11-28:
Re-opening this: I had refrained from updating Home Assistant since my original post, however, it was about time to update to the latest version.

I cleared out the old Integrations as they no longer worked, an followed this guide from the start.
I'm now having issues (Again) getting it to work, however I get the feeling I'm almost there.

When I try to add the integration, I get the following message: **Unknown error occured**
[image]

My log says the following:
`2022-11-28 12:36:57.955 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call system_status
2022-11-28 12:36:57.956 INFO (MainThread) [custom_components.jablotron80] Connecting to JA80 via HID using /dev/hidraw1...
2022-11-28 12:36:57.956 DEBUG (MainThread) [custom_components.jablotron80] Sending startup message
2022-11-28 12:36:58.063 DEBUG (MainThread) [custom_components.jablotron80] Successfully sent startup message
2022-11-28 12:36:58.064 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call mode
2022-11-28 12:36:58.065 INFO (MainThread) [custom_components.jablotron80] JA80CentralUnit(N/A): initializing mode to Unsplit
2022-11-28 12:36:58.066 INFO (MainThread) [custom_components.jablotron80] initializing
2022-11-28 12:36:58.069 DEBUG (ThreadPoolExecutor-5_0) [custom_components.jablotron80] Loop endlessly reading serial`

freschism, 2022-11-28:
Alright - I got it working. The solution was to unplug the USB while HASS was running, Plug it back in (no restart), and check for the new device path. Using this path in the setup, fixed the issue