tahvane1/jablotron80 #8: Unknown message type 198

mazzy89, 2021-05-18:
I get almost every 24 seconds an error message that looks like this:

```
2021-05-18 13:55:52 ERROR (MainThread) [custom_components.jablotron80] Unknown message type 198 with data c6 ff received
```

How could I get what that `c6 ff` data corresponds?

tahvane1, 2021-05-19:
This is probably some kind of keepalive message. What devices you have in your setup?

mazzy89, 2021-05-19:
Yes it would make a lot of sense since it is pushed with regular cadence. I have door sensors, smoke detector, motion sensors

tahvane1, 2021-05-19:
Those probably don't send keepalives. Some "special" keypad?

mazzy89, 2021-05-19:
Yes I have two keypads in the house. One at the main entrance and the second one to the second floor. Two of those

mazzy89, 2021-08-13:
@tahvane1 any thoughts about this? Could you advice how we can whitelist those health checks so to avoid to feed of so much noise the HA logs

closed 2022-01-31 by tahvane1