tahvane1/jablotron80 #10: Get unknown timestamp event when the alarm is armed and trying to disarm with the fob

mazzy89, 2021-05-19:
```sh
Logger: custom_components.jablotron80
Source: custom_components/jablotron80/jablotron.py:1454
Integration: Jablotron 80 (documentation, issues)
First occurred: 6:50:21 PM (2 occurrences)
Last logged: 6:50:27 PM

Unknown timestamp event data=e3 19 05 18 53 44 3e 60 ff

Unknown timestamp event data=e3 19 05 18 53 44 3e 60 ff
```

tahvane1, 2021-05-19:
Could you check from from O-link (events or something like that) what happened at that timestamp? 19 (day), 05 (month),18 (hour), 53 (minutes) , 44 (seconds)

mazzy89, 2021-05-19:
I will check and I will post the result

mazzy89, 2021-05-19:
[image]

These are the actions executed in that timeframe

mattsaxon, 2022-02-01:
So since there are 2 occurences and set and unset are already handled, I suggest this is the ARC message.

mattsaxon, 2022-02-01:
Fixed in PR #32

mattsaxon, 2022-02-02:
I will leave open until a new release is cut with this PR in it so @mazzy89 can verify.

tahvane1, 2022-02-02:
New version was released

mattsaxon, 2022-02-02:
@tahvane1 appreciated. @mazzy89 can you test and we can close.

mattsaxon, 2022-02-17:
Closing due to lack of response, we can reopen if feedback received

closed 2022-02-17 by mattsaxon

mazzy89, 2022-02-20:
thanks. at the moment the issue disapperead.