tahvane1/jablotron80 #11: Getting back vs getting out

mazzy89, 2021-05-24:
Jablotron once is going to be armed it shows in the display the state to go out versus the state to getting back home.

These are two very important states that would be very helpful to automate properly the system.

Would be possible to map those into the logic?

tahvane1, 2021-07-29:
Hi, If I remember correctly integrations already notices these states, but there is no corresponding states in HA. It probably could be possible to have custom functionality if you need to automate something based on these events though...

mattsaxon, 2022-01-03:
At the moment, the "Arming" state is never shown, at least not in my system. It goes straight to "Armed". I've submitted a PR to fix this for @tahvane1 to review. See https://github.com/tahvane1/jablotron80/pull/14

When returning home, the system should be in a "Pending" state, I haven't checked if that works correctly as my system is not setup with an entry delay.

closed 2022-01-31 by tahvane1