tahvane1/jablotron80 #12: Tried to arm the system but the sytem gets back to the disarmed status

mazzy89, 2021-05-29:
Once I try to arm the system, the system immediately returns to the disarmed state. 
Here the logs

```
2021-05-29 20:18:29 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:29 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-05-29 20:18:29 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 00 00 01 00 00 1e 46 ff
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-05-29 20:18:30 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 25.0 to 75.0
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 00 00 01 00 00 28 75 ff
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-05-29 20:18:30 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 75.0 to 100.0
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 00 00 01 00 00 1e 46 ff
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-05-29 20:18:30 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 100.0 to 75.0
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-05-29 20:18:30 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 00 00 01 00 00 0a 26 ff
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-05-29 20:18:31 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 75.0 to 25.0
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-05-29 20:18:31 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:18:32 ERROR (MainThread) [custom_components.jablotron80] Unknown message type 198 with data c6 ff received
2021-05-29 20:18:32 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 00 00 01 00 00 0a 26 ff
2021-05-29 20:18:32 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
[... 885 lines cut ...]
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2021-05-29 20:19:25 INFO (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 25.0 to 50.0
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 00 00 01 00 00 0a 26 ff
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active
[... cut ...]
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2021-05-29 20:19:25 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
```

mazzy89, 2021-05-29:
seems like the system get into a loop state.

tahvane1, 2021-05-31:
This status message appears regularly like many times per second. It is shown in logs as this wireless strength (or something like that) is changing. (duplicates are filtered out).

closed 2021-05-31 by tahvane1

mazzy89, 2021-05-31:
I've fixed the issue by disconnecting and reconnecting the system from the cable.