tahvane1/jablotron80 #17: JA-80T functionality

phrachovina, 2022-01-18:
I have a JA-80T cable that doesn't work with this. However, I am able to perform beta testing and provide the necessary logs for debugging
[jablotron.log](https://github.com/tahvane1/jablotron80/files/7892304/jablotron.log)
.

mattsaxon, 2022-02-01:
Thanks, so looking at the logs, it looks like it is reading correctly, but not sending messages, is that correct?

So can you see the entities in HA and they are correctly updated? But you can set the system with HA?

phrachovina, 2022-02-01:
No, I click the Add integration of Jablotron80, choose JA-80T cable in the port /dev/ttyUSB0 and the master code, and th process wil end with an error "Could not read settings! Is system disarmed?". No entities was added. I enclose the current logs.
[jablotron2.log](https://github.com/tahvane1/jablotron80/files/7982028/jablotron2.log)

tahvane1, 2022-02-02:
I'm suspecting that command to enter elevated mode is somehow different with this cable.

mattsaxon, 2022-02-10:
Yes, I see what you mean from the logs.

@phrachovina can you confirm that by entering *0 and your master code on the keypad you are able to enter "Service Mode".

Note that this code is the code you should be entering into HA during setup on the integration. If you are not entering that code, then the integration will not install and you get the above error.

mattsaxon, 2022-02-11:
On can you confirm to enter this from keypad, on y or system is *0 follow by SC or MC? And that this works?

phrachovina, 2022-02-11:
Yes, of course, it works. Master code enter maitenance mode and service code service mode, but service code should not be used. By default, the service code is known only to the installer, not to the user.

phrachovina, 2022-02-11:
Ok, here are the packets from Olink when I enter the master codde:
[packets.txt](https://github.com/tahvane1/jablotron80/files/8048586/packets.txt)

mattsaxon, 2022-02-12:
Thanks, can I ask if you've successfully used this integration https://github.com/fwpt/HASS-Jablotron80-T?

mattsaxon, 2022-02-12:
With regard to the trace, I think this may be the received data on serial. Can you do a similar trace, but with the sent data when entering the code from the keypad in OLink, rather than at the physical keypad?

phrachovina, 2022-02-12:
I actually entered the code through Olink's keypad.
I also tried the integration Jabkitron80-T, but unfortunately it doesn't work.

phrachovina, 2022-02-12:
I don't know, how to catch packets from each side. In the Olilnk SW there are colors, so I upload picture of this while I'm sending the master code
[image]

mattsaxon, 2022-02-14:
Thanks, I wasn't aware of that tool, but comparing it to my output, it looks compatible. I'll take a closer look later too.

I've contacted fwpt on the home assistant forum and they have the cable working perhaps you should make contact and see if you can work out what is wrong between you given you have same hardware.

mattsaxon, 2022-03-02:
@phrachovina  can I confirm that you've been changing your master code throughout these logs to be different codes?

phrachovina, 2022-03-02:
I used 3 types of codes - user, master and service code. The service code cannot be used for normal use. As well as the master code, It can be used for initial configuration loading, but for normal use, only the user code is used, which is different for all members of the household.

mattsaxon, 2022-03-02:
I'm referring to the logs that are shown in this thread. I see 4 different codes being used, is that because you've changed you ur code? Since only master and service can be used for initial setup, I'd expect to see 2 maximum in the logs, but I see 4.

phrachovina, 2022-03-02:
I used only 3 codes, master, service and my user code.

mattsaxon, 2022-03-03:
Yes I see that now sorry.

mattsaxon, 2022-03-19:
@phrachovina can you try testing with the latest release?

phrachovina, 2022-03-19:
I tried it twice, it ran out for the first time, the devices appeared, but after sending it ended in an error. The second attempt to add integration ended with the same error as before. I enclose logs
[jablotron4.log](https://github.com/tahvane1/jablotron80/files/8309836/jablotron4.log)
.

mattsaxon, 2022-03-19:
Can you try the latest release please (0.25), I've resolved one of the issues you are getting, the first one. 

I'm not sure on the second yet, but i've managed to get mine setup, its definitely unreliable at the moment, but I can usually get mine setup with 3 attempts with the latest code. If you get it setup its a bit more reliable, but certainly not perfect. I want to see if you are having same situation as me, I suspect you are.

(Note I haven't got a JA-80T cable, I've lashed up something similar by hand!)

phrachovina, 2022-03-19:
Same situation - Could not read settings! Is system disarmed? Log included
[jablotron5.log](https://github.com/tahvane1/jablotron80/files/8309924/jablotron5.log)
.

phrachovina, 2022-03-22:
Great, it looks like it works. My devices have been loaded, I haven't named them yet, I hope it will change later. I'm not home now, so I won't fully try it until this afternoon. Sending current logs for review
[jablotron6.log](https://github.com/tahvane1/jablotron80/files/8323294/jablotron6.log)
.

mattsaxon, 2022-03-25:
No errors or warnings, looks good to me

closed 2022-03-25 by mattsaxon

phrachovina, 2022-03-25:
Yes, great job, thanks very much. 👍