tahvane1/jablotron80 #34: Unable to arm

FraBoCH, 2022-02-07:
I used (and slightly contributed) to the old integration made by Matt and I'm switching to this one.
Unfortunately, I'm unable to arm the system from home assistant (from the UI or direct service call).
To be more precise, it switch correctly to "arming" state (even on the physical keypad) and then after about 3 seconds go back to disarmed state.
Log shows only an "unknown activity received"

Logger: custom_components.jablotron80
Source: custom_components/jablotron80/jablotron.py:1643
Integration: Jablotron 80 (documentation, issues)
First occurred: 6 février 2022, 21:50:27 (102 occurrences)
Last logged: 22:00:32

Unknown activity received data=ed 51 44 00 09 00 00 1e 27 ff
Unknown activity received data=ed 51 44 00 09 00 00 00 77 ff
Unknown activity received data=ed 51 44 00 09 04 00 32 6c ff
Unknown activity received data=ed 51 44 00 09 04 00 1e 51 ff
Unknown activity received data=ed 51 44 00 09 04 00 00 01 ff

mattsaxon, 2022-02-07:
As per the code, this is an activity of type 0x44, which is not recognised.

I haven't received any of these myself in the testing that I've conducted, buy my system does set/unset correctly.

I suspect we will need a fuller log with at least info and perhaps debug messages logged, can you obtain us one please?

Note, I will have a new commit ready in next few days which should recognise a few more states, so you may choose to wait a few days for that.

FraBoCH, 2022-02-08:
I think the unknown activity is not directly involved in this issue. Here is the full debug log. We see that the arm command is properly sent and we get as expected the arming state for zone A (I tested with a arm home). But then immediatly after there is a "Zone A action disarm" occurring. 

`2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Sending new command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Command sent b'\x00\x02\x01\x8f'
2022-02-08 19:06:23 INFO (ThreadPoolExecutor-0_0) [custom_components.jablotron80] confirmation for command Command name=keypress *HIDDEN* received
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Sending new command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Command sent b'\x00\x02\x01\x82'
2022-02-08 19:06:23 INFO (ThreadPoolExecutor-0_0) [custom_components.jablotron80] confirmation for command Command name=keypress *HIDDEN* received
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Sending new command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Command sent b'\x00\x02\x01\x80'
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Message of type KeyPress received 8f ff
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Message of type Beep received a0 ff
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Message of type KeyPress received 82 ff
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Message of type Beep received a1 ff
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Message of type Event received e3 08 02 19 06 0d 3f 35 ff
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Last Event: 08 02 19 06 0d 3f
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] JablotronDevice function call active
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] Zone A action arming
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call by
2022-02-08 19:06:23 INFO (MainThread) [custom_components.jablotron80] JablotronZone(1): by changed from code id=0,reaction=Admin,name=Master,code1=****,active=False to Device id=63,model=None,reaction=None,tampered=False,battery_low=Falsename=device_63,active=True
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-08 19:06:23 INFO (MainThread) [custom_components.jablotron80] JablotronZone(1): status changed from Disarmed to Arming
2022-02-08 19:06:23 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2022-02-08 19:06:23 INFO (MainThread) [custom_components.jablotron80] Zone A status changed from Disarmed to Arming
2022-02-08 19:06:23 INFO (MainThread) [custom_components.jablotron80] Date=2022-02-08 19:06:00,event_type=13,source=63
2022-02-08 19:06:23 INFO (ThreadPoolExecutor-0_0) [custom_components.jablotron80] confirmation for command Command name=keypress *HIDDEN* received
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Sending new command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Command sent b'\x00\x02\x01\x85'
2022-02-08 19:06:23 INFO (ThreadPoolExecutor-0_0) [custom_components.jablotron80] confirmation for command Command name=keypress *HIDDEN* received
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Sending new command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Command sent b'\x00\x02\x01\x84'
2022-02-08 19:06:23 INFO (ThreadPoolExecutor-0_0) [custom_components.jablotron80] confirmation for command Command name=keypress *HIDDEN* received
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Sending new command Command name=keypress *HIDDEN*
2022-02-08 19:06:23 DEBUG (ThreadPoolExecutor-0_0) [custom_components.jablotron80] Command sent b'\x00\x02\x01\x85'
2022-02-08 19:06:23 INFO (ThreadPoolExecutor-0_0) [custom_components.jablotron80] confirmation for command Command name=keypress *HIDDEN* received
2022-02-08 19:06:24 DEBUG (MainThread) [custom_components.jablotron80] Message of type KeyPress received 80 ff
[... cut ...]
2022-02-08 19:06:29 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-08 19:06:29 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2022-02-08 19:06:29 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status
2022-02-08 19:06:29 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
2022-02-08 19:06:29 DEBUG (MainThread) [custom_components.jablotron80] JablotronZone function call status`

FraBoCH, 2022-02-09:
To compare, here is the log of event when arming the same zone A successfully from the physical keypad: [arm-from-keypad.log](https://github.com/tahvane1/jablotron80/files/8035963/arm-from-keypad.log)

FraBoCH, 2022-02-09:
Ok I found the root cause (not the fix yet): my alarm is configured to allow arming without entering the code (I only press the "A" or "ABC" taste on the physical keypad to arm it.
For some reason, this setting is not correctly fetched by the integration and thus method "is_code_required_for_arm()" returns true. Then HA sends the master code to arm which disarm the system immediately. If I force is_code_required_for_arm() to return False, then everything works as expected.

mattsaxon, 2022-02-10:
My system is set the same way and I dont have the issue you described. However this field is set in the integration setup. I don't think it is retrieved anywhere. Go into HA and Integration and click "Configure" on the Jablotron integration and you will see the setting.

JohnnyM84, 2022-02-11:
i had the same problem before the changes of mattsaxon.
if i press on configure, you have 2 optiosn both are on for me (require code to disarm and require code to arm)
but my system is armed without code, in the alarm control panel i don't see a keypad if the system is disarmed if it is armed i see a keypad. So looks like this setting does not work.

FraBoCH, 2022-02-12:
Hi guys,
Yes I know about the configuration parameter and I've set it to false as I expect: 
[image]
But for me this parameter should be to decide if the code must be entered in HA UI or not when performing the action. There must be an additional parameter to determine if HA must send the master code to the control unit when arming or not (even if I doubt someone actually configures his control to enter the code every time to arm it). 
For now, I achieved the result I want by hard-coding `return False` in the method `is_code_required_for_arm` in jablotron.py on line 1166:
```
	def is_code_required_for_arm(self) -> bool:
		value = self._settings.get_setting(JablotronSettings.SETTING_ARM_WITHOUT_CODE)
		return  value is None or value == False
```

From what I see in the code, the configuration is fetched from the control unit in jablotron.py on line 1760:
```
	if setting_value == 0x01:
		self._settings.add_setting(JablotronSettings.SETTING_ARM_WITHOUT_CODE,True)
	else:
		self._settings.add_setting(JablotronSettings.SETTING_ARM_WITHOUT_CODE,False)
```
I guess my control unit sends slightly different data which make this setting to be wrongly set (or not set at all). We'll need more logging to figure that out what's happening.

FraBoCH, 2022-02-19:
Ok, I have a fix available working on my system which, in essence, simplifies the "code required when arming" logic to only rely on the integration option "Require code to arm". If true: code must be entered in UI by user, if false then no code is sent to the control unit at all.
The only limitation is that the case where someone configured his control unit to require the code to arm but doesn't want to manually enter the code in HA will not be supported. IMHO, this case is a corner case that can be ignored.
@tahvane1 I would like to publish a branch with this fix & then create a pull request, could you please give me the needed permissions ?

mattsaxon, 2022-02-19:
@FraBoCH you shouldn't need permissions to publish a branch. Fork the repo and do a PR against this repo.

I'd be happy with the fix as you describe, but I'm still confused why you need this as I seem to have exactly the same setup as you.

i.e. I don't need a code to be entered to set it and I don't have the code available for entry into the HA interface.

mattsaxon, 2022-02-19:
Can you search you logs for the E6 03 00 which should then have xx xx FF after it?

mattsaxon, 2022-02-19:
OK, further investigation, can you search for the following in the core.config_entries file in config/.store.

There will be an entry as follows;
```
"settings": {
                        "device_require_code_to_arm": true,
                        "device_system_mode": "Partial"
                    },
```

If you change the `true` to `false`, that I think it will just work (with old code)

The debug I mention above is only available a the integration setup time and I don't expect you want to setup the component again. However if you do, we can correct the root cause of your issue.

FraBoCH, 2022-02-19:
Hi @mattsaxon. I'm a bit puzzled by the differences between our systems indeed. I

FraBoCH, 2022-02-19:
OK, I can confirm I have "device_require_code_to_arm" set to true in core.config_entries. I share your opinion about this being the root cause (the initial config somewhat got the wrong value). 
But it's a bit nuclear to start reconfiguring everything from scratch to find the difference in my case ;-). I think my proposed fix in PR #42 will just work for most users.

mattsaxon, 2022-02-19:
The issue with the fix is that is requires users to configure if they need the code or not rather than the code determining it. I agree its a bit nuclear to recofngiure for you, but I'm reluctant to reduce the code's capability because of this issue. Instead I propose to keep the code the same and you patch the config entry.

FraBoCH, 2022-02-19:
I've reverted to release 0.19, fixed the config as proposed and I can confirm it fix the issue ! Thanks a lot for the help !
So it's indeed a viable workaround. Maybe it would be sufficient to add something in the readme for that until further investigation is done ?
You're also right about the config backup, it could be in fact not that time consuming.... I may give it a try someday.

mattsaxon, 2022-02-19:
I agree that whilst this issue is out there a note in the readme will be useful. In the meantime, i will rename this issue to be more explict.

FraBoCH, 2022-02-22:
I just tried setting up again from scratch. The parameter "device_require_code_to_arm" has been correctly set to false. Maybe some previous code change fixed the isssue.
I'm thus closing this issue.

closed 2022-02-22 by FraBoCH

mattsaxon, 2022-02-22:
Thanks for the update. Well
Keep an eye out for other occurrences and reopen if needed