tahvane1/jablotron80 #36: Unrecognised activity coming out of both Service and Maintenance modes

mattsaxon, 2022-02-12:
Out of Service Mode
```
2022-02-12 08:41:57 ERROR (MainThread) [custom_components.jablotron80] Unknown status message status=8 received data=ed 08 01 00 01 00 00 32 23 ff
```

Out of Maint Mode
```
2022-02-12 08:43:12 ERROR (MainThread) [custom_components.jablotron80] Unknown status message status=40 received data=ed 28 02 00 01 00 00 28 45 ff
```

In addition the logs say Service Mode entered and Exited for both
The only difference is that device 127 is for Service and Device 64 for Maint.

```
2022-02-12 08:41:25 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa0, 1 subtle (short) beep triggered
2022-02-12 08:41:26 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa0, 1 subtle (short) beep triggered
2022-02-12 08:41:27 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa0, 1 subtle (short) beep triggered
2022-02-12 08:41:28 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa1, 1 loud (long) beep triggered
2022-02-12 08:41:28 INFO (MainThread) [custom_components.jablotron80] JablotronCode(63): active changed from False to True
2022-02-12 08:41:28 INFO (MainThread) [custom_components.jablotron80] Date=2022-02-12 08:41:00,event_type=Service Mode Entered, 127:device_127
2022-02-12 08:41:28 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xb7, no audible beep
2022-02-12 08:41:29 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xb8, Infinite beeping triggered
2022-02-12 08:41:29 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa7, no audible beep
2022-02-12 08:41:29 INFO (MainThread) [custom_components.jablotron80] JablotronZone(1): status changed from Disarmed to Service
2022-02-12 08:41:29 INFO (MainThread) [custom_components.jablotron80] JablotronZone(2): status changed from Disarmed to Service
2022-02-12 08:41:29 INFO (MainThread) [custom_components.jablotron80] JablotronZone(3): status changed from Disarmed to Service
2022-02-12 08:41:57 ERROR (MainThread) [custom_components.jablotron80] Unknown status message status=8 received data=ed 08 01 00 01 00 00 32 23 ff
2022-02-12 08:41:57 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xb8, Infinite beeping triggered
2022-02-12 08:41:57 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa1, 1 loud (long) beep triggered
2022-02-12 08:41:57 INFO (MainThread) [custom_components.jablotron80] Date=2022-02-12 08:42:00,event_type=Service Mode Exited, 24:Keypad
2022-02-12 08:41:58 INFO (MainThread) [custom_components.jablotron80] JablotronZone(1): status changed from Service to Disarmed
2022-02-12 08:41:58 INFO (MainThread) [custom_components.jablotron80] Zone A status changed from Service to Disarmed
2022-02-12 08:41:58 INFO (MainThread) [custom_components.jablotron80] JablotronZone(2): status changed from Service to Disarmed
2022-02-12 08:41:58 INFO (MainThread) [custom_components.jablotron80] Zone B status changed from Service to Disarmed
2022-02-12 08:41:58 INFO (MainThread) [custom_components.jablotron80] JablotronZone(3): status changed from Service to Disarmed
2022-02-12 08:41:58 INFO (MainThread) [custom_components.jablotron80] Zone C status changed from Service to Disarmed
2022-02-12 08:41:58 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xb8, Infinite beeping triggered
2022-02-12 08:41:58 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xfe, unknown beep(s) triggered
2022-02-12 08:42:39 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa0, 1 subtle (short) beep triggered
2022-02-12 08:42:40 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa0, 1 subtle (short) beep triggered
2022-02-12 08:42:41 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa0, 1 subtle (short) beep triggered
2022-02-12 08:42:42 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa0, 1 subtle (short) beep triggered
2022-02-12 08:42:42 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa1, 1 loud (long) beep triggered
2022-02-12 08:42:42 INFO (MainThread) [custom_components.jablotron80] JablotronCode(0): active changed from False to True
2022-02-12 08:42:42 INFO (MainThread) [custom_components.jablotron80] Date=2022-02-12 08:43:00,event_type=Service Mode Entered, 64:device_64
2022-02-12 08:42:43 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xb7, no audible beep
2022-02-12 08:42:44 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xb8, Infinite beeping triggered
2022-02-12 08:42:44 INFO (MainThread) [custom_components.jablotron80] JablotronZone(1): status changed from Disarmed to Service
2022-02-12 08:42:44 INFO (MainThread) [custom_components.jablotron80] JablotronZone(2): status changed from Disarmed to Service
2022-02-12 08:42:44 INFO (MainThread) [custom_components.jablotron80] JablotronZone(3): status changed from Disarmed to Service
2022-02-12 08:43:12 ERROR (MainThread) [custom_components.jablotron80] Unknown status message status=40 received data=ed 28 02 00 01 00 00 28 45 ff
2022-02-12 08:43:13 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xb8, Infinite beeping triggered
2022-02-12 08:43:13 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xa1, 1 loud (long) beep triggered
2022-02-12 08:43:13 INFO (MainThread) [custom_components.jablotron80] Date=2022-02-12 08:43:00,event_type=Service Mode Exited, 24:Keypad
2022-02-12 08:43:13 INFO (MainThread) [custom_components.jablotron80] JablotronZone(1): status changed from Service to Disarmed
2022-02-12 08:43:13 INFO (MainThread) [custom_components.jablotron80] Zone A status changed from Service to Disarmed
2022-02-12 08:43:13 INFO (MainThread) [custom_components.jablotron80] JablotronZone(2): status changed from Service to Disarmed
2022-02-12 08:43:13 INFO (MainThread) [custom_components.jablotron80] Zone B status changed from Service to Disarmed
2022-02-12 08:43:13 INFO (MainThread) [custom_components.jablotron80] JablotronZone(3): status changed from Service to Disarmed
2022-02-12 08:43:13 INFO (MainThread) [custom_components.jablotron80] Zone C status changed from Service to Disarmed
2022-02-12 08:43:14 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xb8, Infinite beeping triggered
2022-02-12 08:43:14 INFO (MainThread) [custom_components.jablotron80] Keypad Beep: 0xfe, unknown beep(s) triggered
```

tahvane1, 2022-02-16:
"The only difference is that device 127 is for Service and Device 64 for Maint." This could be also code number.

mattsaxon, 2022-02-16:
I've changed to code and 64 is recognised as master but 127 is a 'none'

mattsaxon, 2022-02-17:
Also any ide what devices from 51 to 63 inclusive are? 51 seems to be (also) Control Panel

mattsaxon, 2022-02-22:
Fix in https://github.com/tahvane1/jablotron80/pull/50

closed 2022-02-24 by tahvane1