tahvane1/jablotron80 #37: Suggestion to remove unknown messages from logs

mattsaxon, 2022-02-12:
I've been clearing up my logs, making as many messages understood as possible.

Given we don't know how to validate the last byte of a message and I seem to get a fair amount of garbage messages in my logs still that look corrupted, i.e. a mix of multiple types of messages such as

```
2022-02-12 09:40:57 ERROR (MainThread) [custom_components.jablotron80] Invalid message of type StateDetail received e8 e8 0e 4b ff
2022-02-12 09:40:58 ERROR (MainThread) [custom_components.jablotron80] Invalid message of type Beep received b7 ed 40 00 00 01 00 00 ed 40 ed 40 00 ed 40 00 00 01 00 00 14 76 ff
2022-02-12 09:40:58 ERROR (MainThread) [custom_components.jablotron80] Invalid message of type State received ed 40 00 00 01 00 00 ed 40 00 00 01 00 00 14 76 ff
2022-02-12 09:40:59 ERROR (MainThread) [custom_components.jablotron80] Invalid message of type State received ed 40 00 00 01 00 00 0a ed 40 00 00 01 00 00 0a 26 ff
2022-02-12 09:40:59 ERROR (MainThread) [custom_components.jablotron80] Invalid message of type State received ed 40 00 00 01 00 00 ed 40 00 00 01 00 00 0a 26 ff
```

I suggest we add a configuration item to config.yaml to disable these unknown errors.

That way we wont have messy logs and we can also turn a simple setting on to show them for debug purposes

@tahvane1 what do you think?

tahvane1, 2022-02-12:
I agree

mattsaxon, 2022-02-13:
@FraBoCH thats certainly easier and I can see the value as its likely that we will need a full debug log to address and issues.

@tahvane1 what do you think to @FraBoCH suggestion?

tahvane1, 2022-02-13:
I agree that this is cleaner solution. However it means that we won't get so many issues anymore... ;)
Someday I will find time to decrypt that check digit calculation...

tahvane1, 2022-02-13:
Changed to main branch