tahvane1/jablotron80 #46: Index out of range

mazzy89, 2022-02-20:
Another error I have found coming from this integration is

```
Unexpected error: Traceback (most recent call last): File "/config/custom_components/jablotron80/jablotron.py", line 709, in read_send_packet_loop records_tmp = self._read_data() File "/config/custom_components/jablotron80/jablotron.py", line 661, in _read_data if data[1 + int(size)] == 0xff: IndexError: index out of range
```

Need to find out how to reproduce it but maybe I get it when I arm the alarm through the home assistant integration.

mattsaxon, 2022-04-27:
I have experienced this myself now and have an idea what the issue is

mattsaxon, 2022-05-08:
Fixed in release 0.27

closed 2022-05-08 by mattsaxon