tahvane1/jablotron80 #74: Entering non delayed zone after delayed zone no alarm

JohnnyM84, 2022-02-25:
If i enter a delayed zone and do not disarm the system and walk to a non delayed zone the alarm sounds but in home assistent there is no alarm. The triggered pir also does not change.

If i directly enter a non delayed zone the alarm sounds and home Assistant sees it and the triggered pir is also changeing to triggered, so problem is only with delayed zone.

mattsaxon, 2022-02-25:
Thanks, can you get me a debug log of the issues? 

Would you mind attaching it as opposed to posting in the text.

JohnnyM84, 2022-02-25:
Attatched the log, in it where a few test:
first arm home and disarm to check if the errors where away
second arm away when i was in front of a pir (got briefly an alarm and than got away, the same if i arm to fast after leaving the living room(think i did it twice)
3rd armed away, entering delayed zone and than entering non delayed zone (the issue for this problem)
4rd armed away, entering non delayed zone.

[home-assistant.log](https://github.com/tahvane1/jablotron80/files/8143735/home-assistant.log)

mattsaxon, 2022-02-27:
OK, so the logs are giving me good information to detect this.

Before we proceed, can I understand if this used to work with previous version of software and if I've broken it with recent changes? From what I can see, I expect not, but want to be sure before making more changes.

It does appear that your system is sending different codes to mine (and presumably @tahvane1), but they look similar enough for me to correctly fix with a bit more info.

Can I check what during all this, what your keypad has on it. It looks from the logs as if the warning icon on your panel is fully lit (not flashing) at some points of the testing.

mattsaxon, 2022-02-27:
Reading the logs, here is what I see;

16:25:34 EXIT_DELAY_A
16:26:24 ARMED_A
16:27:20 DISARMED
16:27:46 Sensor activated (5:woonkamer_pir) - though this is during the system being in disarmed state
16:29:09 EXIT DELAY_ABC
16:29:59 ARMED_ABC
16:30:10 ENTRY_DELAY_ABC
16:30:33 DISARMED
16:31:34 ARMED_ABC
16:32:37 Sensor activated (9:Hal_boven_pir)
16:32:42 Alarm Cancelled

It seems the alarm has only gone off once as you state, but when did you think it should have gone off and what sensor was triggering it

mattsaxon, 2022-02-27:
One last thought, if your sensor is set to NEXT_DELAY, then it shouldn't trigger an alarm to my understanding if you have entered via an sensor which is set to DELAYED. I wonder if this could be the issue? Though obviously, if the main siren is sounding and HA doesn't recognise this, there is an issue, but I can't see any data coming from the alarm in the logs to indicate this.... well not with my current understanding of it.

JohnnyM84, 2022-02-28:
> Reading the logs, here is what I see;
> 
> 16:25:34 EXIT_DELAY_A 16:26:24 ARMED_A 16:27:20 DISARMED 16:27:46 Sensor activated (5:woonkamer_pir) - though this is during the system being in disarmed state (I believe that a solid warning triangle meaning "Fault" may be lit on the keypad at this point) 16:28:04 a different, unknown activity 16:28:42 woonkamer active again ("Fault" may be lit again) 16:28:47 the same unknown activity, again 16:29:09 EXIT DELAY_ABC 16:29:59 ARMED_ABC 16:30:10 ENTRY_DELAY_ABC 16:30:33 DISARMED 16:31:34 ARMED_ABC 16:32:37 Sensor activated (9:Hal_boven_pir) 16:32:42 Alarm Cancelled
> 
> The software has only seen the alarm has only go off once as you state, but when in this timeline do you think it should have gone off and what sensor was triggering it?
> 
> Though note that the time of your HA instance and Control Panel are around 32 mins out of sync.

This are 2 tests, the delayed zone that was triggerd was 04:hal_beneden_pir, but this is not even in the log.

The test that i did where as follows (now added the triggerd devices):
- first arm home and disarm to check if the errors where away
- second arm away when i was in front of a pir (got briefly an alarm and than got away, the same if i arm to fast after leaving the living room, **this test was with 05:woonkamer_pir tested it i think 2 times maybe 3**
- 3rd armed away, entering delayed zone and than entering non delayed zone (the issue for this problem), **this was the delayed zone 04:hal_beneden_pir and the non delayed zone 05:woonkamer_pir, when i enter the delayed zone on screen you see only the message turn off no warning**
- 4rd armed away, entering non delayed zone, **this was with 9:Hal_boven_pir**

I had never tested this before so i can not say it has ever worked but i don't think so.
Can you test it yourself? or is your setup different?

Answers questions:
Can I check what during all this, what your keypad has on it. It looks from the logs as if the warning icon on your panel is fully lit (not flashing) at some points of the testing...... this is new code, so I could have misinterpreted it.
**When entering a delayed zone the warning sign is off only turn off in the screen.**

Can I also check that the alarm is full sounding in your first test as from the logs it looks like it beeps for entry delay, but is then unset.
**The first test with arming when in a non delayed zone there was no sounding alarm only short alarm within home assistant and on screen the message detected 05:woonkamer_pir and the warning sign was fully lit not flashing**

As of your last post alarm is always triggerd if you enter an non delayed zone they are configured als instant and not as natural so it's good that the alarm sounds when you walk from a delayed zone to a non delayed zone and do not disarm it.

mattsaxon, 2022-02-28:
A few things for me to check further.

1. Can I ensure your use of the term 'zone' refers to detectors and not actual zones? There are 3 zones in jabloton A, B and C. To which zones do your detectors belong?

2. When the warning is lit on the keypad can you press the '?' Key and tell me what is displayed.

3. Apologies, when i mentioned your 'first test' I was referring to what you now stated as the 3rd test. What confuses me is that I see no information in the logs during this test suggesting that an alarm or detector is triggered. I shall have to review the code later to ensure it is logging all information correctly.

4. Can I confirm which detectors are wireless vs wired, I presume detector 4 is wired and 5 and 9 are wireless?

with regard to testing the setup, I will try to recreate, but there are obviously some setup differences between our systems so may need to ask some more questions until we establish what differences are relevant.

JohnnyM84, 2022-02-28:
1. Can I ensure your use of the term 'zone' refers to detectors and not actual zones? There are 3 zones in jabloton A, B and C. To which zones do your detectors belong?
**Yes i mean detectors, zone a are the natural detectors zone b are the instant detectors.**

2. When the warning is lit on the keypad can you press the '?' Key and tell me what is displayed.
**this is with the second test in display i see than, Active output pgy**

3. Apologies, when i mentioned your 'first test' I was referring to what you now stated as the 3rd test. What confuses me is that I see no information in the logs during this test suggesting that an alarm or detector is triggered. I shall have to review the code later to ensure it is logging all information correctly.
**Yes i don't see the delayed detector in the logs eighter**

4. Can I confirm which detectors are wireless vs wired, I presume detector 4 is wired and 5 and 9 are wireless?
**I have only wired detectors**

JohnnyM84, 2022-02-28:
Correction for the devices in which zone:
Zone a: 1, 2, 3, 7, 8
Zone b: 4, 5, 6, 9, 10

mattsaxon, 2022-02-28:
What is your PGY contact connected to?

JohnnyM84, 2022-02-28:
Good question, if i look in olink and go to settings i see PGY function set on any system part set = PG on
Do you mean that?

mattsaxon, 2022-03-01:
Are you able to access your keypad whilst you walk between zones (i.e. is it remote)?

It seem that there is no update in the logs between 16:30:10 and 16:30:33 where I believe you expect to see an alarm triggered. Furthermore, the event at 16:30:10 doesn't have any detector details associated with it.

During this period (well a repeat of the test), it would be good know what is displayed on the keypad and also if you press ? what additional info is displayed. If the pressing? during this period gives more info, I will be able to automate that in the code.

What is most surprising however is that no alarm is triggered during that period, i.e. the state of the alarm is in state 0x4b, which means ENTRY_DELAY_ABC, so the data says the alarm is not triggered. How long is your entry delay configured for? seems you clear it within 23 seconds, can you test what happens if you let the entry delay expire, i.e. does the HA integration see the alarm in those circumstances?

As part of this, i've identified some logging improvements that may help to determine the issue, but at this point I'm afraid I'm stumped if I'm honest!

JohnnyM84, 2022-03-01:
My entry delay is on de max i think that is 50 or 60 sec.
If i let the entry delay expire than every thing works as aspected.
I have attatched a log from saterday when this happend. you see an antry delay with no triggered device, 45 sec later you see alarm with device garage_contact. in log there is also 1 unknow activity.
Maybe you something with it.
[log.txt](https://github.com/tahvane1/jablotron80/files/8162306/log.txt)

mattsaxon, 2022-03-01:
So is the log above behaving as you would expect? Was the garage contact the device that triggered the entry delay, perhaps with wired devices this behaves differently. I have all wireless detectors.

Sorry for so many questions, but I could really do with a log of what you did at what time instead of having to guess this from the logs, then I can compare the real life to the logs and see if I can fix the issues based on that.

JohnnyM84, 2022-03-01:
No problem.
Yes, this is what I expect the garage contact is triggering the delay, the key pad beeps and after the delay is over the alarm sounds and in ha I have an alarm too. 

It would be nice that you see in ha a contact sensor as open and a pir as triggered when in the delayed mode. But I can live with how it works now. 

But if je walk from an delayed device to a non delayed device and the alarm sounds you would expect to see it in ha. Didn’t test it yet what happens if I test it and wait for the normal delay but it’s not so good for my hearing ;)

mattsaxon, 2022-03-01:
Have you got "Triggered Detector Indication" enabled in OLink?

mattsaxon, 2022-03-01:
Just reconfigured with my only wired detector and have same issue, it doesn't send any messages to keypad on entry delay, so its confirmed that wired and wireless detectors behave differently

mattsaxon, 2022-03-01:
Further testing and looking at control panel logs shows that the alarm in the instant zone is not recorded on the logs, but if I let the delay expire then the alarm is triggered. The logs show the delayed sensor that was first entered, not the instant sensor.

It seems this is the behaviour of the control panel and as such I don't think there is anything I can do in HA integration to rectify this. I am marking as WontFix as a result

JohnnyM84, 2022-03-01:
Oké thx for your time and effort in investigating it.

mattsaxon, 2022-03-01:
Thanks you for the logs and as a result I've submitted some improvements which improve the integration for you and others.

mattsaxon, 2022-03-02:
Thanks, so we have similar enough setup now and I can reproduce your problems. 

Whilst it may give different results I don't recommend my settings unless you have very active animals!

closed 2022-03-04 by tahvane1

mattsaxon, 2022-03-08:
@JohnnyM84 reading the manual today, section 6.14, I note

"If the unconfirmed entrance delay is confirmed by an instant detector it 
will trigger an internal warning (IW) alarm immediately (e.g. an internal 
siren) and if the entrance delay times out then an external alarm (EW) 
will be triggered. "

This is consistent with the behaviour we observe, i.e. the HA integration only triggers when the external siren is triggered.

I'm marking this issue as 'by design' as a result