tahvane1/jablotron80 #83: What is expected behaviour of code entities?

mattsaxon, 2022-03-04:
At the moment, there are codes entities for each user code and master.

There is no service code entity, though I believe there should be as some of the logged messages are relevant for the service code, for example entering elevated model.

Furthermore, the code entities, for my system never become active (my system is never set by code).

Looking at the python code, the intent appears to be that on setting of the system a code becomes active and on unsetting this would be cleared, I'm not sure if that makes sense given a different code could be used to set the system, from unset and this would leave the entities in a strange state.

Can you clarify what the intended behaviour of the code entities are?

tahvane1, 2022-03-06:
My system is very small and currently only one non-admin code is in use. (I think)  I originally  wanted to use code(s) as part of "presence" detector (more like is anyone present), but I have not realized that. 

I agree that it is a bit problematic. If we want to just track code usage then it would be best to just active and (time delayed) clear.  Another option would be to "track person who armed" and then we would need to clear all codes from unset (from any code).

mattsaxon, 2022-03-06:
My proposal would be to change these entities to be the date/time the user was last seen.

tahvane1, 2022-03-07:
sounds reasonable

mattsaxon, 2022-03-10:
Also should add Service Code as an entity.