tahvane1/jablotron80 #100: Assistance needed to test an specific activity

mattsaxon, 2022-03-06:
There is an  activity that my system cannot create

activity 0x16, which creates text of "Triggered detector (3)"

How is this created on your system @tahvane1 and what is displayed on keypad when the activity occurs?

tahvane1, 2022-03-07:
I believe this is not from my system. Code part is from here  #84

mattsaxon, 2022-03-07:
Ah yes, sorry, it is from @johnym84's system

JohnnyM84, 2022-03-08:
[ja80.log](https://github.com/tahvane1/jablotron80/files/8205792/ja80.log)

mattsaxon, 2022-03-08:
The message ( "Triggered Detector(3)" )gets displayed at a couple of times in the log at 12:27:15 and 12:27:45

and the state message "received ed 40 56"...... occurs many times. 

Is there a way for you to narrow down which detector/activity this is from as I see you are triggering 4 different detectors.

JohnnyM84, 2022-03-08:
Have you an idee how i can narrow this down. 
created 2 times an alarm with 2 devices and both had the problem. 
The 4 devices you saw where devices from a zone that was not armed I think.

mattsaxon, 2022-03-09:
I think I'll change the code to auto do a query to see what it helpless if you can test it

JohnnyM84, 2022-03-09:
Is good, hopefully we can see it then in the logging.

mattsaxon, 2022-03-09:
committed to main as part of https://github.com/tahvane1/jablotron80/pull/104

@JohnnyM84, please retest and see what this yields

JohnnyM84, 2022-03-10:
updated to the lasted main, i have one time the error.
added the relative part of the log.
[ha.log](https://github.com/tahvane1/jablotron80/files/8222805/ha.log)

JohnnyM84, 2022-03-10:
And a log when i make an alarm, i saw nothing special in the message box within lovelace then an alarm message.
[ha2.log](https://github.com/tahvane1/jablotron80/files/8222897/ha2.log)

mattsaxon, 2022-03-10:
Patch made

mattsaxon, 2022-03-10:
Worked it out, it is when multiple detectors are triggered. The screen displays "triggered det." still, but there are multiple ones to see if you press query multiple times

JohnnyM84, 2022-03-10:
Oké, so if I walk from one room to another and I trigger multiple pirs that you get the triggered detector (3) message.

mattsaxon, 2022-03-10:
yep, I have this reproduced now. Just tidying up messaging and will patch again

JohnnyM84, 2022-03-10:
Good find, but I saw the message too if I trigger an alarm and only triggered 1 device.

JohnnyM84, 2022-03-10:
Tested it on a place where I can trigger 2 pirs at once and I can confirm that I now get the message of the triggered detector multiple.
Did not test alarm but that must be the siren and licht that is on when there is an alarm.

mattsaxon, 2022-03-10:
Hmmm ok, pity on you getting it with one only, Can you monitor and see if you can work it out when exactly it occurs. Now I get it too, I can check aswell. I will close this for now pending us having an issue to track. Thanks for help.

closed 2022-03-10 by mattsaxon