tahvane1/jablotron80 #109: Error adding integration

Lirik-JuKi, 2022-03-13:
Hello,
can not add the integration.
[image]

`This error originated from a custom integration.

Logger: custom_components.jablotron80
Source: custom_components/jablotron80/jablotron.py:1726
Integration: jablotron80 (documentation, issues)
First occurred: 11:41:37 (2 occurrences)
Last logged: 11:49:40

Tampering alarm, 5:device_5, 11:41 Sun 13 Mar
Tampering alarm, 5:device_wired_5, 11:49 Sun 13 Mar`
[image]

mattsaxon, 2022-03-13:
The above log doesn't look like its is the full log and the times don't match, i.e. the log is from 11:35 and the errors from 11:41-11:49.

The tampering issues should not cause any problem with adding the integration, however it wouldn't be a bad idea to remove them to clean the logs up!

Please can you supply a log from during the adding of the integration and i will investigate.

Lirik-JuKi, 2022-03-13:
I was removed all jablotron integration, after install as new.
same result.
[image]
[image]
[Jablotron log.txt](https://github.com/tahvane1/jablotron80/files/8239382/Jablotron.log.txt)

mattsaxon, 2022-03-13:
ok, thanks for log, I can see an issue, but its not obvious what it is. Have you used the integration before successfully? If so, can I ask you to try with a previous release and tell me if it is this new release that has the issue? Release 0.14 is the one before I made any changes, so that would be a good one to test.

Lirik-JuKi, 2022-03-14:
This is my first time using the integration. I tried version 0.14, everything is fine there. Integration added.

mattsaxon, 2022-03-14:
Ok, thanks, I will retest setup with new version, but you should be able to upgrade to latest one now the install is done.

mattsaxon, 2022-03-19:
Resolved in release 0.25

mattsaxon, 2022-03-25:
Closing due to lack of response. Reopen if recurrence

closed 2022-03-25 by mattsaxon