tahvane1/jablotron80 #112: Unique Ids contain "too much uniqueness"

mattsaxon, 2022-03-18:
Since the unique IDs contain the serial port name, this means that the system creates duplicate IDs when the serial ports changes.

Since it is unlikely that a house will have 2 alarm systems, this uniqueness is unnecessary IMO and creates issues during debugging or changing of system configuration.

There are a couple of options I can think of;

1. Remove the serial ports from the unique name (this would be a breaking change)
2. Add a system name to the config that overrides the serial name (this would not be a breaking change, but would be a bit more complex to code)

mattsaxon, 2022-03-18:
This issue was found when fixing issue #17 as OI had to reconfigure my system and actually used a remote serial with name "socket://192/168/0/8:23?logging=True".

This creates a set of duplicate IDs and as you can see from how specific a remote name is creates many each time you change IP, port or even debugging info

tahvane1, 2022-03-21:
Maybe option to config to NOT use serial as unique name?