tahvane1/jablotron80 #116: binary_sensor.ja_80k_control_panel sometimes problem

JohnnyM84, 2022-03-23:
Sometimes when i arm the system the binary_sensor.ja_80k_control_panel goes from OK to Problem. This only goes away when i restart home assistant.
Yesterday evening @ 22:44:56 the status was prblem again, i looked in the logging but can't see why.
attatched the logging maybe @mattsaxon can see wat the problem is
[ha.log](https://github.com/tahvane1/jablotron80/files/8330538/ha.log)
.

mattsaxon, 2022-03-23:
Haven't looked at logs, but this happens (by design) when there is an alarm or fault that is attributed to the control panel, eg low battery, tamper alarm etc. there is one alarm which I find strange that does get attributed to control panel, which is unconfirmed alarms.

mattsaxon, 2022-03-23:
Suggest you also look at the Jablotron logs around that time.

JohnnyM84, 2022-03-23:
i haved looked in the logs but can't find a reason why it gives a problem, thats why uploaded the logging from the time of the problem, maybe someone else can see the problem in there.

JohnnyM84, 2022-03-24:
Tonight i got an false alarm (Tamper alarm, 52:Keypad), this was the first time i got it with this integration.
I had it with the old integration of @mattsaxon too. Can this be related with the problem or with this integration?
From this message in the log it looks like so:
WARNING (MainThread) [custom_components.jablotron80] Tampering alarm, 63:Device on line, 01:01 Thu 24 Mar
Attatched some logging of the event.
[tamper.log](https://github.com/tahvane1/jablotron80/files/8339189/tamper.log)

mattsaxon, 2022-03-24:
I don't think it can be the integration, particularly as you got it with both old and new.

To have the control panel read problem, neither of these messages would have triggered it, but you do seem to have a problem with tampering.

The fact that you have messages from 2 different devices, 63 and 52 is worrying. Particularly because 63 is your Ja-82t cable and I didn't even think them have a Tamper sensor.

I'd try to see if there is a pattern and if this is happening regularly, if removing the integration stops it happening.

I've not had any such issues to compare with.

JohnnyM84, 2022-03-24:
I live now 3 years in this house and the first year i never had this issue. After the first year i started to use the integration and i have this problem 2 till 4 times a year now.
So it's not that much (but anoying in the middle of the night) but it looks for me related.
Do i need to open an other issue because it has nothing to do then with the binary_sensor.ja_80k_control_panel that gives as state problem.

mattsaxon, 2022-03-24:
Let me review the logs when I get a chance to advise. I've into been able to interact via phone this week, so haven't been able to check

JohnnyM84, 2022-03-24:
No problem there is no rush

mattsaxon, 2022-03-25:
OK, looks like 2 issues, the control panel one looks like a bug in the code. The partial set doesn't have a device associated with it, so it defaults to the control panel, which registers as "problem". How did you set the alarm at '2022-03-22 22:44:56'

mattsaxon, 2022-03-25:
The 'false' alarms are a different issue, but I'm not sure I can accept this as a bug in the software, the logs clearly show the tampering message coming from the control panel. So the software appears to be working correctly. You have some actual tampering (loosely attached keypad?) or a hardware issue as far as i can tell.

JohnnyM84, 2022-03-25:
> OK, looks like 2 issues, the control panel one looks like a bug in the code. The partial set doesn't have a device associated with it, so it defaults to the control panel, which registers as "problem". How did you set the alarm at '2022-03-22 22:44:56'

I set the alarm with a scene with a google nest speaker.
In the Scene is the code state: armed_home

JohnnyM84, 2022-03-25:
> The 'false' alarms are a different issue, but I'm not sure I can accept this as a bug in the software, the logs clearly show the tampering message coming from the control panel. So the software [...]

I have 2 keypads both wired and screwed into the wall and they do not move.
Difficult problem to find I think. Strange thing it’s almost always around the same time as it happens 1 o clock at night. 
but good to know that it is not related with the integration, thx for looking into it.

JohnnyM84, 2022-03-28:
I think that I know how to reproduce the problem issue, if I set the alarm with Google in my living room and I triggered the pir in that room that is a non delayed pir than i get the problem. The pir in the living room is not in zone a, zone a is the only zone I arm at night.

mattsaxon, 2022-03-28:
Do you have a log that accompanies you doing this activity. It would be useful if you can note the times you do real activities, e.g. set system, trigger PIR, audible alarm heard etc. etc.

JohnnyM84, 2022-03-31:
It’s in the first ha.log @ 22:44:40 I trigger woonkamer_pir then I arm with google and then error comes on 22:44:55

mattsaxon, 2022-04-01:
Thanks, I see it!

mattsaxon, 2022-05-08:
hopefully fixed in release 0.27

closed 2022-05-08 by mattsaxon

JohnnyM84, 2022-05-08:
Thanks, I’m going to test it. 
I see there are much setup improvements, is it advised to readd the integration?

mattsaxon, 2022-05-08:
There is no requirement to rerun setup, it's been tested in both scenarios.

I wouldn't recommend reading as you will have to reenter you preferred device names.