tahvane1/jablotron80 #117: No way to disarm only a zone in partial mode

mazzy89, 2022-03-27:
My Jablotron is set in partial mode and i do not have a way to disarm only the zone A which corresponds to my ground floor.

mattsaxon, 2022-03-27:
Mine is also set in partial mode.

I'm not sure what you mean, that you cannot disarm only Zone A.

My understanding of the system is that in this mode, you can only set A or AB or ABC in this mode. This is a standard Jablotron feature.

To control Zone independently, I think this is called Split Mode?

mazzy89, 2022-03-27:
Ok let’s assume to set A. This would mean to set the alarm only in the A zone. Did I say it right?

mazzy89, 2022-03-27:
To elaborate my use case is that I would like to disarm only the zone A when the entire home zones A,B,C are all set.

mattsaxon, 2022-03-27:
> Ok let’s assume to set A. This would mean to set the alarm only in the A zone. Did I say it right?

To do that, use "Arm Home" on the lovelace control

> To elaborate my use case is that I would like to disarm only the zone A when the entire home zones A,B,C are all set.

That is not possible when in Partial Mode. You would need to setup your alarm in Split mode (which I don't have experience of, but check section 6.23 of the Control Panel manual)

mazzy89, 2022-03-27:
Right then it makes sense. Thanks for the discussion

closed 2022-03-27 by mattsaxon