tahvane1/jablotron80 #129: USB port problem

HAPSagan, 2022-05-05:
Hello
I had trouble adding the integration and did not succeed when I entered: "/ dev / serial / by-id / usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0" When I used: "/ dev / ttyUSB1" it worked fine. Unfortunately, this has meant that the integration sometimes stops working when the alarm is assigned to another USB port. My question is if it is possible to change this setting and try with "/ dev / serial / by-id / usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0" again without having to reinstall the integration. Or is there another good solution to my problem?
/Peter

tahvane1, 2022-05-05:
Hello Peter,
Unfortunately not see #112 

Erkka

mattsaxon, 2022-05-05:
Well there is a hack to not require a reinstall, but it will create duplicate entities at the moment due to the issue referenced.

How often is this occurring?

HAPSagan, 2022-05-05:
Sometimes several times a day unfortunately. I have not been able to figure out what triggers it.

mattsaxon, 2022-05-05:
Ok, so no solution immediately apparent for this to me, well apart from adding a remote device into the chain, which is what I use. I’ll have a think though to see if I can work out anything else.

HAPSagan, 2022-05-05:
Thanks. Then I'll wait a while. What does it mean to - "add a remote device to the chain, which is what I use"?
If I were to remove and add the integration again - would that lead to duplicate entities?

HAPSagan, 2022-05-05:
On the internet I found a tip regarding another device I have, RFXtrx433XL USB transceiver, which meant adding:
#Rfxtrx
rfxtrx:
   usb_path: / dev / serial / by-id / usb-RFXCOM_RFXtrx433XL_DO5OP3OW-if00-port0
to configuration.yaml. Would it be possible to make a similar solution for the alarm?

mattsaxon, 2022-05-05:
So here's the things I'd like you to try, please try using the /dev/serial/by-path/... as well as the "by-id" path you mention above.
Assuming it doesn't work, I'd like a debug log of the issue occurring so I can see the problem occurring.
You can do this without removing you old integration as the integration does support multiple instances.
In terms of removing and adding, yes this should work, but since this issue is occurring multiple times per day, I'm not sure that really helps you... though you could potentially try as a hack having the integration installed against both USB0 and USB1, but you will then have to deal with duplicate entities.

mattsaxon, 2022-05-05:
> "add a remote device to the chain, which is what I use"

This relates to the explanation here https://github.com/tahvane1/jablotron80#remote though crafting you own solution is not without complexities, lets see what the above suggestion yield first

HAPSagan, 2022-05-06:
When I try to add the integration with the "/dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0" address I get the same error which I got last time "Unknown error occurred"

Core log:
2022-05-06 09:05:48 ERROR (ThreadPoolExecutor-4_0) [custom_components.jablotron80] Unexpected error: Traceback (most recent call last):
  File "/config/custom_components/jablotron80/jablotron.py", line 746, in read_send_packet_loop
    records = self._read_data()
  File "/config/custom_components/jablotron80/jablotron.py", line 725, in _read_data
    data = self._connection.read_until(b'\xff')
  File "/usr/local/lib/python3.9/site-packages/serial/serialutil.py", line 663, in read_until
    c = self.read(1)
  File "/usr/local/lib/python3.9/site-packages/serial/serialposix.py", line 595, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

mattsaxon, 2022-05-06:
And does the by-path equivalent also give same error?

HAPSagan, 2022-05-06:
The "/dev/ttyUSB1" integration is up and running with no issues today.
This error was generated when I tried to add the "/dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0"
along with the ""/dev/ttyUSB1"

mattsaxon, 2022-05-06:
Is there no /Dev/serial/by-path equivalent on your system then?

HAPSagan, 2022-05-06:
I have:
Conbee 2
/dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.4.1:1.0
Zwave Aeotec ZW090 Gen5
/dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.4.4:1.0
RFXCOM_RFXtrx433XL
/dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.1:1.0-port0
Jablotron 80
/dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.2:1.0-port0

mattsaxon, 2022-05-06:
And when you add the integration using /dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.1:1.0-port0
Jablotron 80 what do you get?

HAPSagan, 2022-05-06:
Conbee
ttyACM0
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2401728-if00
Delsystem:
tty
Enhetens sökväg:
/dev/ttyACM0
ID:
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2401728-if00
Attribut:
DEVLINKS: >-
  /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2401728-if00
  /dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.4.1:1.0
DEVNAME: /dev/ttyACM0

Zwave
ttyACM1
/dev/serial/by-id/usb-0658_0200-if00
Delsystem:
tty
Enhetens sökväg:
/dev/ttyACM1
ID:
/dev/serial/by-id/usb-0658_0200-if00
Attribut:
DEVLINKS: >-
  /dev/serial/by-id/usb-0658_0200-if00
  /dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.4.4:1.0
DEVNAME: /dev/ttyACM1

RFX
ttyUSB0
/dev/serial/by-id/usb-RFXCOM_RFXtrx433XL_DO5OP603-if00-port0
Delsystem:
tty
Enhetens sökväg:
/dev/ttyUSB0
ID:
/dev/serial/by-id/usb-RFXCOM_RFXtrx433XL_DO5OP603-if00-port0
Attribut:
DEVLINKS: >-
  /dev/serial/by-id/usb-RFXCOM_RFXtrx433XL_DO5OP603-if00-port0
  /dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.1:1.0-port0
DEVNAME: /dev/ttyUSB0

Jablotron
ttyUSB1
/dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0
Delsystem:
tty
Enhetens sökväg:
/dev/ttyUSB1
ID:
/dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0
Attribut:
DEVLINKS: >-
  /dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0
  /dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.2:1.0-port0
DEVNAME: /dev/ttyUSB1

mattsaxon, 2022-05-06:
I'm out my depth here with what your answer even means...... I was of the execration that you would be able to connect the integration using the by-id, the by-path or the ttyUSBx paths. However I can't test this as I don't have a JA-80T.

HAPSagan, 2022-05-06:
Now I'm out my depth :-)
Can you try to explain to me what you want me to test again?
What I tried this morning was to add my alarm with:
/dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0
with my Jablotron 80
/dev/ttyUSB1
Already running.

mattsaxon, 2022-05-06:
Add the integration again and specify the path with this format /dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.2:1.0-port0

HAPSagan, 2022-05-06:
Do I have to remove the integration I already have installed first?

HAPSagan, 2022-05-06:
I tried without removing the one I have and get "Unknown error"
Core log says:
2022-05-06 10:34:32 ERROR (ThreadPoolExecutor-2_0) [custom_components.jablotron80] Unexpected error: Traceback (most recent call last):
  File "/config/custom_components/jablotron80/jablotron.py", line 746, in read_send_packet_loop
    records = self._read_data()
  File "/config/custom_components/jablotron80/jablotron.py", line 725, in _read_data
    data = self._connection.read_until(b'\xff')
  File "/usr/local/lib/python3.9/site-packages/serial/serialutil.py", line 663, in read_until
    c = self.read(1)
  File "/usr/local/lib/python3.9/site-packages/serial/serialposix.py", line 595, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

HAPSagan, 2022-05-07:
Hello again,
Now I have added the integration again with the address /dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.1:1.0-port0.
To be able to do that, I had to deactivate the old integration. Then when I had the new integration installed, it said that the devices were disabled and that I therefore could not use them. Then I tried to activate the first integration and then it seems to work. Unfortunately, I have too little knowledge to understand what this means. Will the entities work with the new integration if the first one goes down or will it make them unreachable? Does this mean that I should uninstall both and then install with the new /dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.1:1.0-port0 again or should I try to have it like this. I think if one integration becomes inactive and the other works - how does the entity work then?

HAPSagan, 2022-05-07:
After a little more research, it may have been the control panel itself that could not be reached as it seems to have received the same entity's Id. The others seem to have got new ones such as device 40 and device_40_2. However, the question remains partly then in view of the control panel. I tried renaming the first control panel to see if the new one would work then, but unfortunately not. It seems that the last control panel added can not be used. I may have to remove both integrations and just add it with the new address or what do you think?

mattsaxon, 2022-05-08:
I'm not quite sure what has happened here, but yes now you've shown it can work using the "by-path" approach, you will need to remove both integrations and reinstall. From what you've said I imagine it should also work with "by-id" also in this situation.

I suspect the old issues you were facing was due to some reliability issues with the general setup process and I've have released a new version to address this and other items. Release 0.27

HAPSagan, 2022-05-08:
I have now removed both the old integrations and tried to add a new.
First with
/dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0
and when that failed, with
/dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.1:1.0-port0
When that also failed I tried with the 
/dev/ttyUSB1
Which also failed
Another thing I noticed was that max number of devices now was 30 instead of 50 earlier. I have more than 40 on my instance.
I have attached the logs from all three tries below.

My last thing to do was to use my backup that had two installations, and now it works as before my attempts to uninstall and reinstall.

/dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0

2022-05-08 11:51:43,904 - custom_components.jablotron80 - DEBUG - JA80CentralUnit function call system_status
2022-05-08 11:51:43,905 - custom_components.jablotron80 - INFO - Connecting to JA80 via Serial using /dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0...
2022-05-08 11:51:43 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call system_status
2022-05-08 11:51:43 INFO (MainThread) [custom_components.jablotron80] Connecting to JA80 via Serial using /dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0...
2022-05-08 11:51:43,908 - custom_components.jablotron80 - DEBUG - JA80CentralUnit function call mode
2022-05-08 11:51:43,909 - custom_components.jablotron80 - INFO - JA80CentralUnit(N/A): initializing mode to Unsplit
2022-05-08 11:51:43 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call mode
2022-05-08 11:51:43 INFO (MainThread) [custom_components.jablotron80] JA80CentralUnit(N/A): initializing mode to Unsplit
2022-05-08 11:51:43 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 94, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 219, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 137, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 206, in post
    return await super().post(request, flow_id)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 62, in wrapper
    result = await method(view, request, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 260, in async_configure
    result = await self._async_handle_step(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 335, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
  File "/config/custom_components/jablotron80/config_flow.py", line 92, in async_step_user
    cu = JA80CentralUnit(None, self._config, None)
  File "/config/custom_components/jablotron80/jablotron.py", line 1441, in __init__
    if options[CONFIGURATION_QUIETEN_EXPECTED_WARNINGS]:
TypeError: 'NoneType' object is not subscriptable

/dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.1:1.0-port0

2022-05-08 11:54:02 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
[... 48 lines cut ...]
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 219, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 137, in handle
    result = await result
[... cut ...]
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 335, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
  File "/config/custom_components/jablotron80/config_flow.py", line 92, in async_step_user
    cu = JA80CentralUnit(None, self._config, None)
  File "/config/custom_components/jablotron80/jablotron.py", line 1441, in __init__
    if options[CONFIGURATION_QUIETEN_EXPECTED_WARNINGS]:
TypeError: 'NoneType' object is not subscriptable
2022-05-08 12:58:11 ERROR (SyncWorker_4) [spotipy.client] Max Retries reached

mattsaxon, 2022-05-08:
Please can you attach the logs as individual files rather than pasted inline.

Number of devices meaning has changed to say 'number of wired devices'. It autodetects wireless devices.

HAPSagan, 2022-05-08:
/dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0

2022-05-08 11:51:43,904 - custom_components.jablotron80 - DEBUG - JA80CentralUnit function call system_status
2022-05-08 11:51:43,905 - custom_components.jablotron80 - INFO - Connecting to JA80 via Serial using /dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0...
2022-05-08 11:51:43 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call system_status
2022-05-08 11:51:43 INFO (MainThread) [custom_components.jablotron80] Connecting to JA80 via Serial using /dev/serial/by-id/usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0...
2022-05-08 11:51:43,908 - custom_components.jablotron80 - DEBUG - JA80CentralUnit function call mode
2022-05-08 11:51:43,909 - custom_components.jablotron80 - INFO - JA80CentralUnit(N/A): initializing mode to Unsplit
2022-05-08 11:51:43 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call mode
2022-05-08 11:51:43 INFO (MainThread) [custom_components.jablotron80] JA80CentralUnit(N/A): initializing mode to Unsplit
2022-05-08 11:51:43 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
resp = await request_handler(request)
File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
resp = await handler(request)
File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 94, in forwarded_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 219, in auth_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 137, in handle
result = await result
File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 206, in post
return await super().post(request, flow_id)
File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 62, in wrapper
result = await method(view, request, *args, **kwargs)
File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
result = await self._flow_mgr.async_configure(flow_id, data)
File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 260, in async_configure
result = await self._async_handle_step(
File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 335, in _async_handle_step
result: FlowResult = await getattr(flow, method)(user_input)
File "/config/custom_components/jablotron80/config_flow.py", line 92, in async_step_user
cu = JA80CentralUnit(None, self._config, None)
File "/config/custom_components/jablotron80/jablotron.py", line 1441, in init
if options[CONFIGURATION_QUIETEN_EXPECTED_WARNINGS]:
TypeError: 'NoneType' object is not subscriptable

HAPSagan, 2022-05-08:
/dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.1:1.0-port0

2022-05-08 11:54:02 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
resp = await request_handler(request)
File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
resp = await handler(request)
File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 94, in forwarded_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 219, in auth_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 137, in handle
result = await result
File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 206, in post
return await super().post(request, flow_id)
File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 62, in wrapper
result = await method(view, request, *args, **kwargs)
File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
result = await self._flow_mgr.async_configure(flow_id, data)
File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 260, in async_configure
result = await self._async_handle_step(
File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 335, in _async_handle_step
result: FlowResult = await getattr(flow, method)(user_input)
File "/config/custom_components/jablotron80/config_flow.py", line 92, in async_step_user
cu = JA80CentralUnit(None, self._config, None)
File "/config/custom_components/jablotron80/jablotron.py", line 1441, in init
if options[CONFIGURATION_QUIETEN_EXPECTED_WARNINGS]:
TypeError: 'NoneType' object is not subscriptable

HAPSagan, 2022-05-08:
/dev/ttyUSB1

2022-05-08 12:57:58,755 - custom_components.jablotron80 - DEBUG - JA80CentralUnit function call system_status
2022-05-08 12:57:58,755 - custom_components.jablotron80 - INFO - Connecting to JA80 via Serial using /dev/ttyUSB1...
2022-05-08 12:57:58 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call system_status
2022-05-08 12:57:58,761 - custom_components.jablotron80 - DEBUG - JA80CentralUnit function call mode
2022-05-08 12:57:58,761 - custom_components.jablotron80 - INFO - JA80CentralUnit(N/A): initializing mode to Unsplit
2022-05-08 12:57:58 INFO (MainThread) [custom_components.jablotron80] Connecting to JA80 via Serial using /dev/ttyUSB1...
2022-05-08 12:57:58 DEBUG (MainThread) [custom_components.jablotron80] JA80CentralUnit function call mode
2022-05-08 12:57:58 INFO (MainThread) [custom_components.jablotron80] JA80CentralUnit(N/A): initializing mode to Unsplit
2022-05-08 12:57:58 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
resp = await request_handler(request)
File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
resp = await handler(request)
File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 94, in forwarded_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 219, in auth_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 137, in handle
result = await result
File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 206, in post
return await super().post(request, flow_id)
File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 62, in wrapper
result = await method(view, request, *args, **kwargs)
File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
result = await self._flow_mgr.async_configure(flow_id, data)
File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 260, in async_configure
result = await self._async_handle_step(
File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 335, in _async_handle_step
result: FlowResult = await getattr(flow, method)(user_input)
File "/config/custom_components/jablotron80/config_flow.py", line 92, in async_step_user
cu = JA80CentralUnit(None, self._config, None)
File "/config/custom_components/jablotron80/jablotron.py", line 1441, in init
if options[CONFIGURATION_QUIETEN_EXPECTED_WARNINGS]:
TypeError: 'NoneType' object is not subscriptable
2022-05-08 12:58:11 ERROR (SyncWorker_4) [spotipy.client] Max Retries reached

HAPSagan, 2022-05-08:
After going back to my backup I tried to install the integration update and that made my integration stop working.
A copy of the log below

Logger: custom_components.jablotron80
Source: custom_components/jablotron80/jablotron.py:1091
Integration: Jablotron 80 (documentation, issues)
First occurred: 13:26:58 (2 occurrences)
Last logged: 13:27:07

Unexpected error processing record: b'\xed@@\x00\x00}\xed@\x01\x00}@@\x00\x01\x00\x00}\xed'
Unexpected error processing record: b'\xed\x00\x00\x00}@\x00\x01\x00}@'
Traceback (most recent call last):
  File "/config/custom_components/jablotron80/jablotron.py", line 2596, in processing_loop
    self._process_message(record)
  File "/config/custom_components/jablotron80/jablotron.py", line 2455, in _process_message
    message_type = JablotronMessage.get_message_type_from_record(data,packet_data)
  File "/config/custom_components/jablotron80/jablotron.py", line 1128, in get_message_type_from_record
    if not JablotronMessage.check_crc(record):
  File "/config/custom_components/jablotron80/jablotron.py", line 1091, in check_crc
    assert packet[length-1] == 0xff
AssertionError

mattsaxon, 2022-05-08:
Try new release 0.28

mattsaxon, 2022-05-08:
The last issue you have looks like it may be from an incorrect choice of JA-82T vs JA-80T on install. For these logs, can you post the entire debug log and as  a file, not as inline comments. This makes them easier for me to process and keep this chat history concise.

HAPSagan, 2022-05-08:
Can you instruct me how to create this log file?

HAPSagan, 2022-05-08:
The latest in this story is that I have now installed 0.28. The result was now that my original installation via / dev / ttyUSB1 works. The one that was parallel (/dev/serial/by-path/platform-xhci-hcd.1.auto-usb-0:1.1.1:1.0-port0) stopped working so I removed it and at the moment I only have / dev / ttyUSB1 installation.

mattsaxon, 2022-05-08:
> Can you instruct me how to create this log file?

I use the "log viewer" add-in to be able to view logs realtime and then copy from that into a file using notepad.

HAPSagan, 2022-05-08:
What do you think I should try to do next?

HAPSagan, 2022-05-09:
Now I have good news! I tried again this morning to remove the integration and add it with
/ dev / serial / by-id / usb-Silicon_Labs_JABLOTRON_serial_interface_1-if00-port0
and it worked! I have no idea why and what was different this time, but I'm happy and satisfied. Thank you so much for your work and support!
I have a completely different question now. If I want to support your work, how do I do it?

HAPSagan, 2022-05-12:
My alarm panel is working. I can turn on and off zones, but I can not se any status changes of , for example, my door sensor. Is that normal?

HAPSagan, 2022-05-12:
I'm pretty sure this was seen last week. Is it a reinstallation that needs to be done again do you think?

HAPSagan, 2022-05-12:
Forget it! It was me this time. In my desperation to quiet the beep sound when the front door was open, I managed to click in the wrong check box. Sorry!

mattsaxon, 2022-05-29:
> if I want to support your work

Many thanks, I have a page here, as I don't have sponsorship setup on GitHub yet.

https://www.buymeacoffee.com/XTOsBAc

closed 2022-05-29 by mattsaxon