tahvane1/jablotron80 #138: sensors do not switch back to their real state

flaakesk8, 2022-08-24:
HI, thank you very much for this plugin I've literally been waiting for years for an jablotron 80 Plugin :).

but I've one issue I see my sensors e.g. when I open a window it does change in the view to open, but never fails back to closed once the window is closed. Same issue with PIRs and doors no matter if the sensors are wired or RF.

The only time I can convince HASS to switch the sensors to it's real state is, when I restart HASS completely.

I'd appreciate if you could tell me what to do - I run the latest HASS on a raspberry Pi 3 B directly connected with USB to my alarm system and LAN cable to the switch.

thank
Joey

mattsaxon, 2022-08-24:
IIRC, that is by design, at least for wireless sensors. When an alarm is tripped, the intent is it stays tripped, until you disarm the system.

For wired sensors, my recollection is that they close when closed at least for door/window sensors.

To progress this, please can you specify the sensor model you think is behaving incorrectly and I will take a look what it does in my system.

flaakesk8, 2022-09-10:
HI mattsaxon, 
thank you for your quick reply :). 
I have issue with the sensors when the alarm system is not actived (not triggered, no zone A, B, nor C is active) I'm at home and eg would like to track if some windows/doors are open or if I should turn on the light once the PIR sensor tirggers.

I have PIR sensors which are hard wired and also door/windows sensors with are wireless and both of them change their state once to triggered but do not fail box once they are eg. closed.
I you want I can check for the sensor type from Jablotron but I have an Oasis 80.
thank you very much
Joey

flaakesk8, 2022-09-29:
whatever magic happended, it's working now as I want it to :) after some home assistant updates - hence I'll close the request.

closed 2022-09-29 by flaakesk8