tahvane1/jablotron80 #139: Motion sensor trigger

markunca, 2022-09-03:
Hi,
Any chance to have more granular info about when motion sensor triggered some motion? (Per pulse On/Off)
Previously, I was using JA-80 though MQTT and get info for each pulse, which I cannot see in this plugin. So I cannot utilize already these sensors in room for another purpose of triggers.

Thanks

tahvane1, 2022-09-04:
Hi,
I don't have motion sensors attached, but at least door/window detectors already work this way (when alarm is not set). If you could provide logs for triggering motion sensor and resetting it so we could take a look...

Br,
Erkka

markunca, 2022-09-05:
Hi,
I know that sensor is sending it towards JA80, I was measuring it in PINs, here what I had there and planned to replace by this direct integration: [https://github.com/markunca/arduino-jablotron](https://github.com/markunca/arduino-jablotron)

Signal from sensor towards JA80 station is send up for every move (generally for every blink of sensor) and down within second after movement ended. This allowed me to trigger different scenes. Not sure, what you can get from JA80 serial, but this info should flow there.

tahvane1, 2022-09-05:
If it works similarly as other sensors then state in serial "stays" up if alarm is triggered until it has been cleared. So exact behaviour is not probably possible.  But if you could serial trace/logs either from Jablotron O-link directly or from HA logs (as instructed for Raw data in front page of this repo) we could take a look.

closed 2024-03-30 by mattsaxon