tahvane1/jablotron80 #140: [question] Can the integration expose a service to trigger a siren?

raducotescu, 2022-09-06:
I have successfully installed the integration to test its features and one thing I didn’t see is a way to trigger the siren. Can this be achieved on these systems or is the siren fully controlled by the alarm’s control panel, without any access to it from the serial port?

tahvane1, 2022-10-25:
Intergration doesn't support it (for now), but if it is possible to trigger siren from control panel then I believe it could be done via integration as well.

mattsaxon, 2022-10-25:
I achieve something similar with the addition of a custom physical sensor that is actually a smart switch. I then control this via Home Assistant

raducotescu, 2023-01-07:
I actually found a way to do this using the panic alarm. PR coming soon.

commit 64d3a2a3 (2023-01-07): fix #140: Add support for triggering the alarm via the alarm_trigger service

* added support for triggering the alarm using the panic mode
commit dfa49814 (2023-01-07): Merge pull request #148 from raducotescu/issue/140

fix #140: Add support for triggering the alarm via the alarm_trigger service
closed 2023-01-07 by tahvane1