tahvane1/jablotron80 #142: Cannot arm alarm with version 0.29

raducotescu, 2022-09-22:
My alarm is configured in unsplit mode and doesn’t require a code for arming. When trying to arm it, I see the following in the logs:

```
2022-09-22 08:32:20.789 DEBUG (MainThread) [custom_components.jablotron80] Adding command Command name=key sequence *HIDDEN*
2022-09-22 08:32:34.873 DEBUG (MainThread) [custom_components.jablotron80] Message of type State received ed 40 00 00 01 00 00 1e 46 ff
2022-09-22 08:32:34.873 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active2022-09-22 08:32:34.874 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active2022-09-22 08:32:34.874 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active2022-09-22 08:32:34.874 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active2022-09-22 08:32:34.874 DEBUG (MainThread) [custom_components.jablotron80] JablotronLed function call active2022-09-22 08:32:34.874 DEBUG (MainThread) [custom_components.jablotron80] JablotronAlert function call value
2022-09-22 08:32:34.875 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor function call value
2022-09-22 08:32:34.875 DEBUG (MainThread) [custom_components.jablotron80] JablotronSensor(1): value changed from 0.0 to 75.0
2022-09-22 08:32:34.877 DEBUG (MainThread) [custom_components.jablotron80] publishing updates
2022-09-22 08:32:34.877 DEBUG (MainThread) [custom_components.jablotron80] Zone A action disarm
2022-09-22 08:32:34.878 DEBUG (MainThread) [custom_components.jablotron80] Zone B action disarm
2022-09-22 08:32:34.878 DEBUG (MainThread) [custom_components.jablotron80] Zone C action disarm
```

What is wrong? I kept looking through the code and didn’t see anything obvious, except for the fact that in unsplit mode, when the control panel is configured to not require a code, the integration simulates a press on the “C” key of the keypad. However, I think our alarm arms when pressing on the “ABC” key.

Moreover, I tried to reconfigure the control panel and the integration to require a code before arming. The result is the same.

mattsaxon, 2022-09-22:
My system is setup same as yours and works as expected.

I suspect that what has happened for you is that there was a comms issue during initial setup and your system thinks it is in partial or split mode.

To check this, you'd need to edit the setup file, but I can't recall where this is at the moment and am away so cannot check. Perhaps someone else can point you in the right direction until I can verify?

raducotescu, 2022-09-22:
Matt, by system you mean the HA integration or the actual control panel? O-Link shows that the system is configured in unsplit mode.

mattsaxon, 2022-09-22:
I'm referring to the HA integration

raducotescu, 2022-09-22:
Here's what I see in `.storage/core.config_entries`:
```json
{
        "entry_id": "7fb1f51ccda4a2d3106b9ab85579ef5f",
        "version": 1,
        "domain": "jablotron80",
        "title": "Jablotron 80",
        "data": {
          "cable_model": "JA_82T",
          "serial_port": "/dev/hidraw0",
          "password": "<redacted>",
          "number_of_wired_devices": 4,
          "settings": {
            "device_require_code_to_arm": false,
            "device_system_mode": "Unsplit"
          }
}
```

mattsaxon, 2022-09-22:
There's another file with setting derives during initial config. this is where all the sensors are stored. It has additional info in it.

raducotescu, 2022-09-22:
Matt, I found the bug:

https://github.com/tahvane1/jablotron80/blob/b7e9d5833a5572b8d0788d13d2df2d043280f230/custom_components/jablotron80/alarm_control_panel.py#L137-L148

If no code is required, the code will be the empty string. In unsplit mode the `arm` function will then be called only with an empty string and the prefix you hardcoded:

https://github.com/tahvane1/jablotron80/blob/b7e9d5833a5572b8d0788d13d2df2d043280f230/custom_components/jablotron80/jablotron.py#L2574-L2578

The correct solution, according to the Jablotron Control Panel manual as well is to:
* Press key ABC, A or B
* Enter a code (or present a card)
* If the system is partially set (section A is set), and you wish to extend the proportion of the system which is set, press the B or ABC key.

I locally changed the code to send the "C" zone (which is the equivalent of the ABC button, that in my case is the only one that arms the system) when the system is configured in unsplit mode and the code is not required and it worked. Should I send the PR for the `alarm_control_panel.py` file?

mattsaxon, 2022-09-22:
Yes please submit a PR so I can see what you intend.... but I'm not yet convinced there is a bug given mine and your systems seem to be configured the same.

There are 2 similar, but different setting regarding does the control panel need a code and do you want a code entered in HA.

..... it could well be me who is confused on this though ;-)

raducotescu, 2022-09-22:
> There are 2 similar, but different setting regarding does the control panel need a code and do you want a code entered in HA.

Those two are currently in sync when the component is set up and the code doesn't seem to adapt if the user changes the component's configuration. If the control panel doesn't require a code, the value configured in HA is overridden with the empty string. I haven't changed that behaviour in the PR, but I see that you already have an issue that could address this - #43.

raducotescu, 2022-09-22:
PR is at https://github.com/tahvane1/jablotron80/pull/143 (don't know why it didn't get linked here).

commit 883d0e68 (2022-10-06): Merge pull request #143 from raducotescu/issue/142

fix #142: Cannot arm alarm with version 0.29 when no code is required for arming
commit e075a764 (2022-10-06): fix #142: Cannot arm alarm with version 0.29 when no code is required for arming

* when the JA-82K control panel doesn't require a code for arming and
the panel is configured in unsplit mode the arming must be done by
simulating a keypad press on the ABC button; sending an empty code
has no effect
closed 2022-10-06 by tahvane1