tahvane1/jablotron80 #144: wireless sensors not added / recognised

mliebl73, 2022-11-02:
I am using the Jablotron integration since some time and it worked nicely so far.

I now needed to uninstall the integration, and after re-installation the wireless sensors are not recognised. Thea are listed as wired sensors and are marked as unavailable + the following text is displayed in the entity view:

This entity is no longer being provided by the jablotron80 integration. If the entity is no longer in use, delete it in settings.

don't know if I was doing something wrong, but those sensors had been working before.

Also (if that is important), the device names were not imported during the installation, they all showed up as device_wired_...

Thanks for the - besides that - great integration!!

mattsaxon, 2022-11-02:
I changed the way this worked a few releases back. Seems like it may have not worked for you.

Can I suggest you try an older release for the install? Try 0.26 and see if that fixes problem.

hcrohland, 2022-12-21:
@mliebl73 did you select 0 wired devices on setup? 
In that case it didn't detect any wired devices for me at all. II had to set it to at least 1 wired device.

mliebl73, 2022-12-21:
@hcrohland I think I set the wired devices to 50 or so - wired devices were not the issue but the wireless ones

tried in the meantime v0.26 but it is still looking weird (no wireless devices detected)

mattsaxon, 2022-12-21:
Some traces would be useful to track diagnose this.

mliebl73, 2022-12-21:
understood, when I find the time to tinker with it again, I will capture those