tahvane1/jablotron80 #145: Device names lost after reconfiguration

hcrohland, 2022-12-20:
I had the integration configured and it had detected the device types and names.
I had to remove it and recoonfigure it. Now it does not detect the device names any more. Instead it shows the serial numbers. Even older versions do not detect the names.

Do I have dreamt that it detected the device names before?

mattsaxon, 2022-12-20:
Looks a duplicate issue of #144

hcrohland, 2022-12-20:
> Looks a duplicate issue of #144

I do not think so. The devices are detected and available. But it does not get the device names/texts and types. It names the devices like "serial 1234567" instead of "Fire alarm kitchen" 

I cannot imagine that I configured all these names manually before...

mattsaxon, 2022-12-20:
I have done some changes to the name algo, but yes there was no way it knew it was on kitchen before.

I take the type of device from the prefix of the serial number.

So if you have decides that arent recognised, please provide device type and serial number

hcrohland, 2022-12-20:
Strange, how did I get all the names into the system 🤔

Anyways, I have a lot of JA-63S-80 Smoke detectors whose serial numbers all start with 019[6789] and a JA-80Z repeater with serial number 06231...

mattsaxon, 2022-12-20:
When the integration configures, you should get a pop up to edit the names. Sometimes it doesn't occur. Haven't tracked down why that is.

hcrohland, 2022-12-20:
Yes, I get that. But I would have bet that I did not add all these names manually in that dialog box. Strange.

mattsaxon, 2022-12-20:
You could try an older version to see what used to happen.

hcrohland, 2022-12-20:
> You could try an older version to see what used to happen.

I did that with the same result.

> there was no way it knew it was on kitchen before.

Apparently I had entered all the names manually. And now I did that again.
Thanks for your help in clarifying.

closed 2022-12-29 by tahvane1