tahvane1/jablotron80 #146: Binary Sensor states are only updated once

migueldc73, 2022-12-22:
When I open a window or door, the binary_sensor state in HA almost immediately changes to "Open", but then this state stays indefinitely "Open", even when I close the window or door. This happens with both wired and wireless windows/doors.
Same for the motions sensors. Once they trigger to "Detected", they stay like that.

The only thing that triggers the binary sensors back to their correct state in HA is arming and then disarming the alarm. The state is actually updated during disarming.

I use the latest version of HASS

mattsaxon, 2022-12-22:
This was, at least partly, intended behaviour. With this approach you can see easily what triggered an alarm in HA

I agree it could be tidied up.

migueldc73, 2022-12-22:
That's really a shame, because most of my windows and doors have wired sensors connected to the Jablotron. And I wanted to use the state of these sensors (open/closed) for other automations in HASS. And this without having to add wireless sensors to all doors and windows.
Any idea how I could achieve this ?

tahvane1, 2022-12-22:
I think there is issue atm as it used not to work like this. Yes sensors kept their state if alarm was triggered and that stayed as long it was cleared (same behaviour as within Jablotron). Now status stays even when system is not armed. I will investigate when I have time. I had "alert" when door stayed open too long even when system was not armed. That is not working anymore. I will check when I have time..

mattsaxon, 2022-12-22:
Yes I think that would be sensible behaviour.

I suspect I may have broken the unarmed behaviour unintentionally

migueldc73, 2022-12-22:
It would be great if you could fix that !
Then we can use the state of all my windows, doors and motion sensors when in unarmed mode.
Many thanks, you guys are quick responding !

tahvane1, 2022-12-29:
This issue is fixed (at least in my case in) PR #147

migueldc73, 2022-12-30:
> This issue is fixed (at least in my case in) PR #147

AWESOME, thanks !!!

closed 2023-01-06 by tahvane1

mleitnercom, 2023-04-14:
Hi,

first I'd like to thank you for the great job done here - I really like this integration!

Unfortunately in my case the problem described above is still existing. What is maybe interesting: I have one door, which is updated immediately after closing - and then also all other windows / doors are updated.

I would be very glad if that situation could be improved.

I use 0.32 and Home Assistant 2023.4.3

Thank you and best regards Michael