tahvane1/jablotron80 #150: Option to scan all device slots?

filip-van-hoeckel, 2023-01-29:
During installation the user gets prompted to configure the amount of devices connected. But there's no way to know the amount beforehand in combination which sensor is on which slot/ID. How would I go about this?

mattsaxon, 2023-01-29:
If I recall correctly, it. prompts for how many wired  sensor there are, not the total.

mattsaxon, 2023-01-29:
Therefore for my config, I use 4, the standard for a Ja80, but it still detects around 26 total.

filip-van-hoeckel, 2023-01-29:
Are detection results logged?

mattsaxon, 2023-01-29:
Yes there will be, but I don't have them to hand.

What's the issue you are facing? Perhaps if you are having an install I said, you can explain it and post a full log of the events during the problem you are experiencing?

filip-van-hoeckel, 2023-01-29:
I've been trying for over a year to integrate this and found the connection to be working by accident. I just keyed in the amount of sensors I have in my house, but did not realize the sensors being added by default - and which sensor is which. So was wondering how found sensors relate to the amount of sensors I keyed in during setup. And which is which ;-) I hope this makes sense.

mattsaxon, 2023-01-29:
The number of wireless sensors detected is up to 50, this comes back in the data stream. However data for wired sensors doesn't come back so this needs hardcoding using the parameter that you enter.

This behaviour changed during the time you've been attempting to integrate. I changed this because my sensors were at odd numbered slots so I didn't want the unused slots to be detected and create unused entities in HA.

filip-van-hoeckel, 2023-01-29:
Where can I look at the 'datastream'? I figure I need to assess the data in order to establish what is which and hopefully get useful identifiers in order to establish sensor and location.

mattsaxon, 2023-01-29:
If you turn in debug logging you will see all you need to see in the HA logs. Refer to the instruction in the repos main readme

filip-van-hoeckel, 2023-01-29:
Thanks, will pursue this route. Thanks for your prompt response - really hoping I will (finally) get this operational ;-)

mattsaxon, 2023-01-29:
Happy to help you get it operational. I realise that it can be hard to setup on some cases, but for me at least once done so it is rock solid and I have no issues with using it as my primary means to control the alarm

closed 2024-03-30 by mattsaxon