tahvane1/jablotron80 #152: Simplify checksum code

perexg, 2023-04-03:
Hi, the checksum code used in the original Windows application is pretty simple:

```
        def checksum(msg):
            sum = 0x7f
            for byte in msg:
                for j in range(8):
                    sum <<= 1
                    if (byte & 0x80) == 0x80:
                        sum += 1
                    if (sum & 0x80) == 0x80:
                        sum ^= 0xa3
                    byte <<= 1
            return sum

        # binary message without 0xFF
        if len(msg) > 1 and checksum(msg) != 0:
            raise SomeCheckSumError
```

perexg, 2023-04-03:
The whole https://github.com/tahvane1/jablotron80/commit/1968736d2e500143b295cbc28cf04eea23b2e1ca can be replaced. I am using this code in my own project for many years.

mattsaxon, 2023-04-03:
That's a whole lot simpler! But can't we check it is as complete..... when you say 'original windows application', what do you refer? Can you reference it please?

mattsaxon, 2023-04-03:
Quite possibly, I lifted the code from somewhere I found an implementation.

Given your example is O-Link its 'better' IMO.

My code does currently seem to get some CRC issues in normal running. I don't know if others see these issues reported in their logs