tahvane1/jablotron80 #153: Window/door sensors are only reported as closed when every window/door has been closed

mleitnercom, 2023-04-22:
Hi,

first I'd like to thank you for the great job done here - I really appreciate this integration!

Unfortunately I am experiencing a behaviour which might be related to issue #146:

In my case the window and door sensors are updating them as open immediately after they were opened. But they update themselfes to closed not before EVERY window/door has been closed.

I would be very glad if that situation could be improved so that the sensors update themselfes immediately after their physical state has changed.

I use 0.32 and Home Assistant 2023.4.4

Thank you and best regards Michael

tahvane1, 2023-05-29:
I have to double check this but as far as I remember this is limitation of Jablotron.

dartapt, 2023-06-11:
O have the same problem.

migueldc73, 2023-06-27:
I have exactly the same problem and it worked well (doors were flagged as closed as soon as they got closed even if others were still open) until a few weeks ago. So this is definitely related to the HA upgrade as I don't remember there was a recent upgrade of this custom integration ?

My feeling is also that it's related to issue #146

migueldc73, 2023-08-09:
This problem is still going on and it's very annoying, because it still triggers a door or window as open even when I closed it, only until all windows are closed.

The problem occurred a few releases ago too and it was solved here: 
https://github.com/tahvane1/jablotron80/pull/147/commits/a0a9d478a002af07b73fde3efc498a7e58b1024c

Obviously this fix does not work anymore, so I hope someone can solve this.

mattsaxon, 2023-08-09:
Have you gone back and tried different releases and confirmed it working in any? That would be helpful to identify when any reversion occurred

migueldc73, 2023-08-09:
So, I am sure it was solved in v0.31 (see (https://github.com/tahvane1/jablotron80/issues/146:) , but when I now went back to v0.31, we have the same behaviour as in v0.32 (or v0.30 for that matter)

So the problem with the "sticky sensors" does not see to come from any change in this integration, but a change in HASS that caused this.

Does anybody have any idea why and how ?
Did you use any depreciated HA functions in the code that caused a problem with the 2023.6 HA updates ?

Reloading the integration does not work, only restarting HA.

Maybe there is a way to force this integration to reset the sensor status, but based on what trigger ? If it does not sense a door is closed again, it cannot trigger the reset. Bit of a conundrum...

heifisch, 2023-08-11:
Nothing has happened here for a long time, but if there are still people who are interested in a solution to this problem, here is a patch.
For me it works very well with 3 wired devices.
[update_devices.patch.gz](https://github.com/tahvane1/jablotron80/files/12324352/update_devices.patch.gz)

heifisch, 2023-08-11:
here is the patch without code cosmetics.

[update_devices_smal.patch.gz](https://github.com/tahvane1/jablotron80/files/12324870/update_devices_smal.patch.gz)

HAPSagan, 2023-08-11:
Hi
This sounds great! Can anyone give me instructions how to install it?
Regards Peter

heifisch, 2023-08-12:
> Hi This sounds great! Can anyone give me instructions how to install it? Regards Peter

If you don't want to use patch, you can copy the jablotron.py file in the attached zip archive to the "custom_components/jablotron80" directory and replace the existing file.
But make a backup of the original file first.

[jablotron.zip](https://github.com/tahvane1/jablotron80/files/12326614/jablotron.zip)

HAPSagan, 2023-08-12:
Ok. Thank you. I will try that later when I am back at home.
Regards Peter

mattsaxon, 2023-08-13:
Hi @heifisch, would you submit your changes as a PR to this repository so others can benefit easily from the fix?

heifisch, 2023-08-13:
> Hi @heifisch, would you submit your changes as a PR to this repository so others can benefit easily from the fix?

Yes, I will do that.
I still have to learn how to do that though.

mleitnercom, 2023-08-15:
Thank you so much @heifisch !

commit 579cb936 (2024-03-30): fix #153 (The second attempt) Window/door sensors are only reported as closed when every window/door has been closed
commit 5ebd0196 (2024-03-30): fix #153 (The third attempt) Window/door sensors are only reported as closed when every window/door has been closed
commit 9d632968 (2024-03-30): Revert "fix #153 (The third attempt) Window/door sensors are only reported as closed when every window/door has been closed"

This reverts commit 5ebd0196da7bc0a53e2c8000ccf44bb91025f54f.
commit a58daee8 (2024-03-30): fix #153 (The fourth attempt) Window/door sensors are only reported as closed when every window/door has been closed
closed 2024-03-30 by mattsaxon