tahvane1/jablotron80 #164: Question: Remote support

majo551, 2024-02-16:
The documentation mentions it is possible to use 

> The JA-80T serial cable setup can work with remote serial devices using a device address of 'socket://[ipaddress:socket]', see section at bottom of page for more details. 

However there is no more details on this config to be done later on at the bottom or elsewhere, only possible hardware is mentioned, that could be used.  I found another project where they showed a RP with ser2net implementing serial to tcp communication with JA-80 and HA

[https://medium.com/@jpsfs/home-assistant-jablotron-remote-access-50d5b305e2c7](url)

But the actual remote config in HA is referred back to this project documentation :-) So it is in a cycle and I cannot figure out how to integrate the remote connection to HA using the generic statement  'socket://[ipaddress:socket]': where, how?.

Can you please help give me some more specific hints how I can make the running remote connection in  HA ?
I tried to create a TCP sensor on the  given IP/port where ser2net runs, it seems HA can see it:
[image]
[image]

,, but the JA-80T integration offers only the USB port setup:
[image]

Thanks for your help. 
Sorry if I did not get it from the docs, it is my first time trying to use Serial to TCP to HA connection.

majo551, 2024-02-17:
Thanks, maybe for dummies like me, the docs should say a real  example ; I tried few version until the right one came :-)
socket://196.10.20.2:2000

Anyway, all worked fine, although I was little bit unprepared to see the values read from the jablotron were all in an undefined default names. I understand that the names of each entities come  without their set names (?)  but I was not prepared to have all  the   entities to be set  and their real names and types during integration, so I had to go to the alarm first to copy all settings down and then start again the integration and set each entity  one by one in HA from scratch. Also this piece of information is missing in the docs and would help me to get ready before integrating to HA. 
Anyway all works now, good work man!

Final question is how to turn off debug logs completely ? There are so many logs coming from the integration, I want to see them only when an issue comes out not to make so many writes ... I did not put the debug logs element to the config  as mentioned in the docs. And despite I get all the debugs logs 

[image]

Thank you for the work once again.

mattsaxon, 2024-03-30:
There's a setting in the integration options for verbose logging, are you aware of that?

mattsaxon, 2024-03-30:
Readme updated with example

closed 2024-03-30 by mattsaxon