tahvane1/jablotron80 #166: USR-TCp232-304

thebadboyltz, 2024-03-29:
I have just installed the USR box on my JA card (terminal A and B).
In home assistant I cannot connect to the USR.
Am I doing something wrong?

the port is 5220 and not 10000 as in the photo

THANKS

[image]
[image]
[image]
[image]

thebadboyltz, 2024-03-30:
please help is welcome

mattsaxon, 2024-03-30:
I don't have "LINK" or "Similar RFC2217" selected on the "Serial Port" screen

mattsaxon, 2024-03-30:
you should also have selected "JA-80T" on the screen when you setup in HA, not JA-82T

thebadboyltz, 2024-03-31:
really thank you I can connect.
But there is one weird thing, my keyboards are going crazy, they repeatedly display all my sensors as if the keyboard command # was pressed.

To disarm the alarm I must systematically enter the password, or can we configure it by default to avoid having to type it?

last thing, the integration uses a lot of CPU resources at times.
[image]

mattsaxon, 2024-03-31:
The looping is a duplicate of #167 

The high CPU is a side effect of this.

closed 2024-03-31 by mattsaxon