tahvane1/jablotron80 #167: Wireless keypad spammed when multiple wireless sensors are active

ondrejmirtes, 2024-03-30:
Spinning this issue off the issue: https://github.com/tahvane1/jablotron80/issues/163

My wireless keypad is being spammed by the integration and cycles through all active sensors every second or so. This means that when the alarm is armed but multiple windows are opened, user interacting with the keypad is not able to disarm the alarm.

ondrejmirtes, 2024-03-30:
Possibly connected to https://github.com/tahvane1/jablotron80/issues/168.

ondrejmirtes, 2024-03-30:
The cause of this behaviour is that with multiple active detectors, this branch is always executed, at least once a second:

https://github.com/tahvane1/jablotron80/blob/af8560cc8f3eacaa83989bf6deff80441382df1c/custom_components/jablotron80/jablotron.py#L2162-L2164

When I comment that out, I don't get details about the 2nd+ detectors but the keyboard does not get spammed.

So we should probably focus on how to reduce the spamming while still getting details about multiple detectors.

ondrejmirtes, 2024-03-30:
FWIW with these lines commented, at least the integration is now usable for me - I can see the state and arm/disarm from my phone, and the keypad is usable too.

So solving these problems would be nice, but it's no longer urgent for me.

ondrejmirtes, 2024-03-31:
Alright, try this:

1) Keep one window open (one sensor active)
2) Arm the alarm
3) Press "?" on the keyboard, maybe even multiple times. Mine shows "Active detector" and then "Active output PGY".
4) In my case the keypad starts cycling (pressing "?" by HA integration) on its own again.
5) When I comment out https://github.com/tahvane1/jablotron80/blob/af8560cc8f3eacaa83989bf6deff80441382df1c/custom_components/jablotron80/jablotron.py#L2137-L2138 (a branch for a single detector), the cycling stops.

mattsaxon, 2024-03-31:
Yes I can see a clear issue here, and how it gets into a loop.

When it "hits" the device query button, it is waiting for /a4/ff messages, but this only occurs once it has cycled through all the info on the keypad, equivalent of pressing the button multiple times.

And when the messages coming back from this trigger yet more queries, hence the loop occurs.

mattsaxon, 2024-04-01:
The issue is a result of a few things;

1. The code in process_state is overly complex, it needs a refactor
2. The integration is trying to do something the alarm control panel itself is not designed to do, that is to provide a real time view of what devices are activated. When using the control panel, this can only be achieved by cycling through the status using the "?" key
3. There are quite a lot of different configurations out there, and each of us has a different config
             a. PGX/Y setup
             b. Wired sensors
4. This integration has not test suite

To resolve this problem whilst also achieving what @heifisch in #153 has set out to do needs quite a lot of effort!

For now, I recommend people turn off PGX/PGY and ensure there are no faults in their system. This will result in a stable configuration.