tahvane1/jablotron80 #168: Not able to arm/disarm alarm reliably when multiple wireless sensors are active

ondrejmirtes, 2024-03-30:
Spinning this issue off the issue: https://github.com/tahvane1/jablotron80/issues/163

Possibly connected to: https://github.com/tahvane1/jablotron80/issues/167

When multiple sensors are active, I'm not able to reliably arm/disarm the alarm from HA. This is how arming looks like in the UI:

https://github.com/tahvane1/jablotron80/assets/104888/38709051-6d30-4466-90dd-114f904c7683

And this is how disarming looks like in the UI:

https://github.com/tahvane1/jablotron80/assets/104888/c4069a34-4cea-477e-b0ae-f54de446aa50

mattsaxon, 2024-03-30:
Are you never able to arm/disarm in this situation? I've tried a couple of times and even when #167 is in effect and I can arm & disarm.

That said I'm using a direct connection with JA-82T cable, what are you using?

ondrejmirtes, 2024-03-30:
I'm using this device (https://www.pusr.com/products/rs232-rs485-to-wifi-converters-usr-w610.html) connected to A+B in the control panel, and added remotely to HA via `socket://`.

I also observed the same behaviour when I connected a Raspberry Pi with JA-82T cable to the control panel and connected it with HA through usbip.

mattsaxon, 2024-03-30:
OK, and it always happens, i.e. you can never arm/disarm the system when it is in this state?

ondrejmirtes, 2024-03-30:
I learnt that it works okay for a while after I restart HA.

Also it looks like the disarming/arming eventually happens, albeit after a few minutes.

I'll try to figure out some way to reproduce it.

mattsaxon, 2024-03-30:
Given what you are saying, seems same behaviour as mine and this is basically a side effect of #167 , I propose we close

ondrejmirtes, 2024-03-30:
I'm not sure about it. When I close all windows and the keypad is no longer spammed, the arming/disarming is still slow/delayed after restart. I'm working on a reproduction to see when it starts happening.