tahvane1/jablotron80 #170: options

thebadboyltz, 2024-04-01:
Hello, what do these options mean, no description appears for me.
[image]
[image]

tahvane1, 2024-04-02:
Sorry. French is not supported. Switch your language to english or provide language file for fr https://github.com/tahvane1/jablotron80/blob/main/custom_components/jablotron80/translations/en.json

thebadboyltz, 2024-04-03:
no problem, I would never have thought of that. I will translate the file for you, it may be useful to the French-speaking community.

last question then I'll stop bothering you, I don't use night mode and present mode, how can I remove these 2 modes from this view

[image]

thebadboyltz, 2024-04-05:
please 👍👍

tahvane1, 2024-04-07:
I think this is more homeassistant thing and not this addon. I'm using custom dashboard for mine and I have limited visible states like this
         entity: alarm_control_panel.jablotron_control_panel_abc
                name: Control panel
                states:
                  - arm_away
You can find the documentation here
https://www.home-assistant.io/dashboards/alarm-panel/

thebadboyltz, 2024-04-07:
unless I am mistaken, the ID which comes from the integration to HA this name: alarm_control_panel.jablotron_control_panel_a_ab_abc, therefore it is not possible from a card to modify this, you have to do it from your code but I cannot find not or. 
I don't need present or night mode and sometimes my wife gets it wrong. 
I just need to turn it off and away but the only place it is is like in my picture