tahvane1/jablotron80 #185: JA-80 connection via TCP stopped working after 2025.1.0 upgrade on Home Assistant Operating System

vervoto1, 2025-01-03:
Hi,

After upgrading to 2025.1.0 for some reason my Jablotron 80 integration via TCP (connected through rasb-pi) stopped working. The jablotron integration didn't show an error but all devices were unavailable. When reloading the integration it gave an error. restarting either the rasb-pi and/or homeassistant didnt solve it so I turned on debug logging.. the result is below.

Downgrade to 2024.12.5 fixed the issue immediately, works again like a charm so I assume some python modules got updated and that's what causing it. As i'm running The Home Assistant Operating System I cannot fiddle with the python modules any more.

2025-01-03 22:27:14.104 ERROR (ThreadPoolExecutor-1_0) [custom_components.jablotron80] Unexpected error: %s
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/serial/urlhandler/protocol_socket.py", line 171, in read
    raise SerialException('socket disconnected')
serial.serialutil.SerialException: socket disconnected
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/custom_components/jablotron80/jablotron.py", line 749, in read_send_packet_loop
    records = self._read_data()
  File "/config/custom_components/jablotron80/jablotron.py", line 896, in _read_data
    data = self._connection.read_until(b'\xff')
  File "/usr/local/lib/python3.13/site-packages/serial/serialutil.py", line 663, in read_until
    c = self.read(1)
  File "/usr/local/lib/python3.13/site-packages/serial/urlhandler/protocol_socket.py", line 178, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: socket disconnected
2025-01-03 22:27:14.104 ERROR (ThreadPoolExecutor-1_0) [custom_components.jablotron80] Unexpected error: %s
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/serial/urlhandler/protocol_socket.py", line 171, in read
    raise SerialException('socket disconnected')
serial.serialutil.SerialException: socket disconnected
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/custom_components/jablotron80/jablotron.py", line 749, in read_send_packet_loop
    records = self._read_data()
  File "/config/custom_components/jablotron80/jablotron.py", line 896, in _read_data
    data = self._connection.read_until(b'\xff')
  File "/usr/local/lib/python3.13/site-packages/serial/serialutil.py", line 663, in read_until
    c = self.read(1)
  File "/usr/local/lib/python3.13/site-packages/serial/urlhandler/protocol_socket.py", line 178, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: socket disconnected
2025-01-03 22:27:14.105 ERROR (ThreadPoolExecutor-1_0) [custom_components.jablotron80] Unexpected error: %s
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/serial/urlhandler/protocol_socket.py", line 171, in read
    raise SerialException('socket disconnected')
serial.serialutil.SerialException: socket disconnected
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/custom_components/jablotron80/jablotron.py", line 749, in read_send_packet_loop
    records = self._read_data()
  File "/config/custom_components/jablotron80/jablotron.py", line 896, in _read_data
    data = self._connection.read_until(b'\xff')
  File "/usr/local/lib/python3.13/site-packages/serial/serialutil.py", line 663, in read_until
    c = self.read(1)
  File "/usr/local/lib/python3.13/site-packages/serial/urlhandler/protocol_socket.py", line 178, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: socket disconnected
2025-01-03 22:27:14.106 ERROR (ThreadPoolExecutor-1_0) [custom_components.jablotron80] Unexpected error: %s
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/serial/urlhandler/protocol_socket.py", line 171, in read
    raise SerialException('socket disconnected')
serial.serialutil.SerialException: socket disconnected
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/custom_components/jablotron80/jablotron.py", line 749, in read_send_packet_loop
    records = self._read_data()
  File "/config/custom_components/jablotron80/jablotron.py", line 896, in _read_data
    data = self._connection.read_until(b'\xff')
  File "/usr/local/lib/python3.13/site-packages/serial/serialutil.py", line 663, in read_until
    c = self.read(1)
  File "/usr/local/lib/python3.13/site-packages/serial/urlhandler/protocol_socket.py", line 178, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: socket disconnected
2025-01-03 22:27:14.107 ERROR (ThreadPoolExecutor-1_0) [custom_components.jablotron80] Unexpected error: %s
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/serial/urlhandler/protocol_socket.py", line 171, in read
    raise SerialException('socket disconnected')
serial.serialutil.SerialException: socket disconnected
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/custom_components/jablotron80/jablotron.py", line 749, in read_send_packet_loop
    records = self._read_data()
  File "/config/custom_components/jablotron80/jablotron.py", line 896, in _read_data
    data = self._connection.read_until(b'\xff')
  File "/usr/local/lib/python3.13/site-packages/serial/serialutil.py", line 663, in read_until
    c = self.read(1)
  File "/usr/local/lib/python3.13/site-packages/serial/urlhandler/protocol_socket.py", line 178, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: socket disconnected

denhuff, 2025-01-04:
whitch version of ja-80 do you use? i run an old qone, and now i have the same problem. wondering if upgrade to version 36+ would work

vervoto1, 2025-01-04:
good point, thanks for the clue! I ran 0.14 for some reason.. didnt see any updates in HACS until I manually added the repository again because I moved from hosted on debian python venv to HAOS.. for some reason still running ancient version. Upgraded to jablotron 80 v0.38 (which reports as 0.35 on the dashboard page for some reason but shows as 0.38 in HACS) and upgraded to HA 2025.1.0 and all working!

denhuff, 2025-01-05:
thanks, gonna try it also

closed 2025-01-05 by tahvane1