tahvane1/jablotron80 #186: deprecated constant which will be removed in HA Core 2026.1

blavak68, 2025-01-05:
I don't know if you noticed, but after upgrade on 2025.1.0 i have this new two warning in the log:

first one:
```
Logger: homeassistant.components.vacuum
Zdroj: helpers/deprecation.py:222
Integrácia: Vysávač (dokumentácia, problémy)

STATE_ERROR was used from purei9, this is a deprecated constant which will be removed in HA Core 2026.1. Use VacuumActivity.ERROR instead, please report it to the author of the 'purei9' custom integration
STATE_IDLE was used from purei9, this is a deprecated constant which will be removed in HA Core 2026.1. Use VacuumActivity.IDLE instead, please report it to the author of the 'purei9' custom integration
STATE_PAUSED was used from purei9, this is a deprecated constant which will be removed in HA Core 2026.1. Use VacuumActivity.PAUSED instead, please report it to the author of the 'purei9' custom integration
STATE_RETURNING was used from purei9, this is a deprecated constant which will be removed in HA Core 2026.1. Use VacuumActivity.RETURNING instead, please report it to the author of the 'purei9' custom integration
STATE_CLEANING was used from purei9, this is a deprecated constant which will be removed in HA Core 2026.1. Use VacuumActivity.CLEANING instead, please report it to the author of the 'purei9' custom integration
```

and second:

```
Logger: homeassistant.helpers.frame
Zdroj: helpers/frame.py:324

Detected that custom integration 'purei9' is setting state directly. Entity None (<class 'custom_components.purei9.vacuum.PureI9'>) should implement the 'activity' property and return its state using the VacuumActivity enum at custom_components/purei9/vacuum.py, line 32: async_add_entities(. This will stop working in Home Assistant 2026.1, please create a bug report at https://github.com/Ekman/home-assistant-pure-i9/issues
```

tahvane1, 2025-01-05:
Wrong integration

closed 2025-01-05 by tahvane1

blavak68, 2025-01-05:
ouu sorry ist my mistake