tahvane1/jablotron80 #189: Detected blocking call to import_module with args ('serial.urlhandler',)

majo551, 2025-03-15:
There is an error  in logs that was told to be raised with the developer. Copying it from logs:

`2025-03-15 09:40:56.085 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to import_module with args ('serial.urlhandler',) inside the event loop by custom integration 'jablotron80' at custom_components/jablotron80/jablotron.py, line 904: self._connection = serial.serial_for_url( (offender: /usr/local/lib/python3.13/site-packages/serial/__init__.py, line 74: importlib.import_module(package_name)), please create a bug report at https://github.com/tahvane1/jablotron80/issues
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once
    handle._run()
  File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 171, in async_setup_component
    result = await _async_setup_component(hass, domain, config)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 467, in _async_setup_component
    await asyncio.gather(
  File "/usr/src/homeassistant/homeassistant/setup.py", line 469, in <genexpr>
    create_eager_task(
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 45, in create_eager_task
    return Task(coro, loop=loop, name=name, eager_start=True)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 782, in async_setup_locked
    await self.async_setup(hass, integration=integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 548, in async_setup
    await self.__async_setup_with_context(hass, integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 637, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/jablotron80/__init__.py", line 54, in async_setup_entry
    cu = JA80CentralUnit(hass, entry.data, entry.options)
  File "/config/custom_components/jablotron80/jablotron.py", line 1504, in __init__
    self._connection.connect()
  File "/config/custom_components/jablotron80/jablotron.py", line 904, in connect
    self._connection = serial.serial_for_url(

2025-03-15 09:40:56.090 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to import_module with args ('.protocol_socket', 'serial.urlhandler') inside the event loop by custom integration 'jablotron80' at custom_components/jablotron80/jablotron.py, line 904: self._connection = serial.serial_for_url( (offender: /usr/local/lib/python3.13/site-packages/serial/__init__.py, line 75: handler_module = importlib.import_module(module_name, package_name)), please create a bug report at https://github.com/tahvane1/jablotron80/issues
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 213, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 154, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 707, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 678, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.13/asyncio/base_events.py", line 2033, in _run_once
    handle._run()
  File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 171, in async_setup_component
    result = await _async_setup_component(hass, domain, config)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 467, in _async_setup_component
    await asyncio.gather(
  File "/usr/src/homeassistant/homeassistant/setup.py", line 469, in <genexpr>
    create_eager_task(
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 45, in create_eager_task
    return Task(coro, loop=loop, name=name, eager_start=True)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 782, in async_setup_locked
    await self.async_setup(hass, integration=integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 548, in async_setup
    await self.__async_setup_with_context(hass, integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 637, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/jablotron80/__init__.py", line 54, in async_setup_entry
    cu = JA80CentralUnit(hass, entry.data, entry.options)
  File "/config/custom_components/jablotron80/jablotron.py", line 1504, in __init__
    self._connection.connect()
  File "/config/custom_components/jablotron80/jablotron.py", line 904, in connect
    self._connection = serial.serial_for_url(

`

z-jun, 2025-04-04:
This issue causes HA core does not start when serial connection can not be established. Possible solution may be use of async serial module, or make function with infinite connection loop asynchronous.

tahvane1, 2025-04-05:
Yes. Fixing this is on my TODO list, but unfortunately ATM that list is quite long...

tahvane1, 2025-08-26:
can you test branch https://github.com/tahvane1/jablotron80/tree/async

majo551, 2025-08-31:
I just updated to the the latest version , since then I have regular issues with the integration failing and being stopped.  The error in logs says:

> This error originated from a custom integration.

Logger: custom_components.jablotron80
Source: custom_components/jablotron80/jablotron.py:746
integration: Jablotron 80 (documentation, issues)
First occurred: 11:32:07 (652711 occurrences)
Last logged: 11:49:13

Unexpected error in packet loop
Traceback (most recent call last):
  File "/config/custom_components/jablotron80/jablotron.py", line 746, in read_send_packet_loop
    records = await self._read_data()
              ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/jablotron80/jablotron.py", line 902, in _read_data
    data = await asyncio.to_thread(self._connection.read_until, b"\xff")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/threads.py", line 25, in to_thread
    return await loop.run_in_executor(None, func_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.13/site-packages/serial/serialutil.py", line 663, in read_until
    c = self.read(1)
  File "/usr/local/lib/python3.13/site-packages/serial/urlhandler/protocol_socket.py", line 155, in read
    raise PortNotOpenError()
serial.serialutil.PortNotOpenError: Attempting to use a port that is not open